tinygo-org/tinygo #3599: reflect.StructOf is missing

adranwit, 2023-03-25:
When trying compile code with reflect.StructOf I see
: undefined: reflect.StructOf

- creating go struct on the fly can not be achieved without  reflect.StructOf call

dgryski, 2023-03-25:
Correct.  TinyGo does not have the ability to create types at runtime at the moment.  The best we can do is stub the call to panic if called, and hope nobody calls it..

adranwit, 2023-03-25:
Would it be possible to add it, I am happy to contribute (any direction to kick start it), I see that currently for struct type definition the following is used structNamesSidetable, structTypesSidetable, I would guess that the latter are used for unnamed types  it is possible at runtime to operated on structTypesSidetable ?

dgryski, 2023-03-26:
This is actually a large chunk of work.