tinygo-org/tinygo #3608: sha512 HardFault

hugolgst, 2023-03-27:
Hello,

I am building on ATSAMD21E18A and for some reason when I use sha512.New my program crashes and prints a single "f" in the logs. I tried using sha256 and it works perfectly. 

I identified that the problem happens when I am summoning the Sum function, here: https://cs.opensource.google/go/go/+/refs/tags/go1.20.2:src/crypto/sha512/sha512.go;l=282
I cannot seem to get more information from this issue, anyone got an idea? Thank you in advance

–hugo

dgryski, 2023-03-28:
Does this reproduce for native code or any other platform?

hugolgst, 2023-03-28:
@dgryski  The code works well on my machine but does not either on the board I am adding to TinyGo nor the Raspberry Pico.

soypat, 2023-04-02:
I am struggling to reproduce the issue on the RP2040 (pico). I've come up with the following:
```go
package main

import (
	"crypto/sha512"
	"time"

	"golang.org/x/crypto/pbkdf2"
)

func main() {
	const last = 4 * 256
	time.Sleep(time.Second)
	println("start program")
	var (
		salt = []byte("add some salt and pepper")
	)

	for i := 0; i < 10000; i++ {
		b := make([]byte, last+i)
		for j := 0; j < last+i; j++ {
			b[j] = byte(i)
		}
		b = pbkdf2.Key(b, salt, 2048, 64, sha512.New)
		println(string(b))
		time.Sleep(time.Second / 4)
	}
}
```

kawaii-clorthax, 2023-04-19:
This is a Go program that generates and prints 10,000 PBKDF2 hashed keys. PBKDF2 is a key derivation function that is used to generate cryptographic keys from a password. In this program, the PBKDF2 function takes a byte slice and a salt as inputs, along with a few other parameters like the number of iterations and the length of the output key. The program generates a byte slice of increasing length on each iteration and hashes it using PBKDF2. The resulting hash is printed to the console, and the program waits for a quarter of a second before continuing to the next iteration.

davecheney, 2026-10-27:
Sorry for the long delay in following up. I tested the sample above with TinyGo 0.44-dev (`b7ec945f`) on a Pico 2 W, limiting the loop to 20 iterations. All completed without a fault, and all 64-byte derived keys matched an independent PBKDF2-HMAC-SHA512 implementation.

This was on RP2350 hardware, rather than the original SAMD21 or RP2040, so it does not establish that the original failure is fixed.

I’m closing this for now because we don’t have a program that reproduces the reported fault. If you still encounter it with a current TinyGo release, please open a new issue with a minimal reproducer, your board or target configuration, and the TinyGo version. Thanks for reporting it.

closed 2026-10-27 by davecheney