tinygo-org/tinygo #3621: reflect.TypeOf inconsistency

adranwit, 2023-03-29:
reflect.TypeOf(T) return inconsistent hashcode when initialized in global block vs within a function:

The following code show the inconsistency issue:

package main

import (
	"fmt"
	"reflect"
)

type Foo struct{}
type Bar struct{}

var types = map[reflect.Type]string{
	reflect.TypeOf(Bar{}): "Bar",
	reflect.TypeOf(Foo{}): "Foo",
}

func main() {
	fooTypeName, ok := types[reflect.TypeOf(Foo{})]
	fmt.Printf("hasFoo: %v key(%p) %v\n", ok, reflect.TypeOf(Foo{}))
	fmt.Printf("fooTypeName: '%v'\n", fooTypeName)
	fmt.Printf("%p\n", reflect.TypeOf(Foo{}))
	printReflectTypeHashCodes(types)
	var rTypes = map[reflect.Type]string{
		reflect.TypeOf(Bar{}): "Bar",
		reflect.TypeOf(Foo{}): "Foo",
	}
	printReflectTypeHashCodes(rTypes)
}

func printReflectTypeHashCodes(types map[reflect.Type]string) {
	for k := range types {
		fmt.Printf("%p\n", k)
	}
}

I would expect that reflect.TypeOf(T) return the same consistent type with the same hashcode

aykevl, 2023-03-30:
This seems to be working well for me, what's the bug?

```
$ go run tmp/issue3621.go
hasFoo: true key(0x1024eec40) %!v(MISSING)
fooTypeName: 'Foo'
0x1024eec40
0x1024eebe0
0x1024eec40
0x1024eebe0
0x1024eec40

$ tinygo run tmp/issue3621.go
hasFoo: true key(0x100229510) %!v(MISSING)
fooTypeName: 'Foo'
0x100229510
0x1002294a0
0x100229510
0x1002294a0
0x100229510
```

adranwit, 2023-03-30:
In my output hasFoo: is false

deadprogram, 2023-05-22:
Reopening, will close on this next release.

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing. Thank you everyone!

closed 2023-06-14 by deadprogram