tinygo-org/tinygo #3631: No serial output on xiao-esp32c3

scottfeldman, 2023-04-01:
I'm not seeing any serial output for xiao-esp32c3.

```
sfeldma@nuc:~/work/tinygo$ tinygo flash -monitor -target xiao-esp32c3 -size short  src/examples/serial/serial.go 
   code    data     bss |   flash     ram
   4502       0    8818 |    4502    8818
esptool.py v4.6-dev
Serial port /dev/ttyACM0
Connecting...
Chip is ESP32-C3 (revision v0.4)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:00:2b:b8
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00001fff...
Compressed 4592 bytes to 3264...
Wrote 4592 bytes (3264 compressed) at 0x00000000 in 0.1 seconds (effective 380.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Connected to /dev/ttyACM0. Press Ctrl-C to exit.
```

I expected to see "hello world!" printed to monitor every second.

No output.

Tried same on xiao board (not xiao-esp32c3) and I see output.

sago35, 2023-04-01:
USB Serial on ESP32-C3 is not available because `USB Serial/JTAG Controller` is not implemented now.

https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started/#resources

scottfeldman, 2023-04-01:
> USB Serial on ESP32-C3 is not available because `USB Serial/JTAG Controller` is not implemented now.
> 
> https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started/#resources

What would it take to add this support?

sago35, 2023-04-01:
I don't know much about it because there is not much documentation.

https://files.seeedstudio.com/wiki/XIAO_WiFi/Resources/esp32-c3_datasheet.pdf

[image]

Perhaps there is a hint in the source code around

https://github.com/espressif/arduino-esp32

It will probably finally be implemented in machine/machine_atsamd51.go, for example.
But first, we need to make sure that code like `xxx.WriteByte('a')` works.

deadprogram, 2023-04-30:
@sago35 you are quite correct.

Further dev info can be found here:

https://github.com/espressif/esp-idf/blob/master/components/soc/esp32c3/include/soc/usb_serial_jtag_reg.h
https://github.com/espressif/esp-idf/blob/master/components/hal/esp32c3/include/hal/usb_serial_jtag_ll.h

aykevl, 2023-04-30:
I think I looked into this once, and IIRC it was very similar to the UART but shows up as a USB-CDC endpoint.

deadprogram, 2024-05-01:
This was already implemented in release v0.31 so now closing. Thank you!

closed 2024-05-01 by deadprogram