tinygo-org/tinygo #3637: -monitor flag is confused with -port flag

scottfeldman, 2023-04-02:
I have two identical raspberry pi picos, /dev/ttypACM0 and /dev/ttyACM1.

When I flash using both -monitor and -port together, the right port is flashed but monitor connects to the wrong port:

```
sfeldma@nuc:~/work/tinygo$ ls /dev/ttyACM*
/dev/ttyACM0  /dev/ttyACM1

sfeldma@nuc:~/work/tinygo$ tinygo flash -monitor -target pico -size short -port /dev/ttyACM0 src/examples/echo/echo.go 
   code    data     bss |   flash     ram
   8548     108    3288 |    8656    3396
Connected to /dev/ttyACM1. Press Ctrl-C to exit.       <-- monitor connected to /dev/ttyACM1

sfeldma@nuc:~/work/tinygo$ tinygo flash -monitor -target pico -size short -port /dev/ttyACM1 src/examples/echo/echo.go 
   code    data     bss |   flash     ram
   8548     108    3288 |    8656    3396
Connected to /dev/ttyACM0. Press Ctrl-C to exit.       <-- monitor connected to /dev/ttyACM0
```

If I get real crazy and add a third pico, nothing works:

```
sfeldma@nuc:~/work/tinygo$ ls /dev/ttyACM*
/dev/ttyACM0  /dev/ttyACM1  /dev/ttyACM2
sfeldma@nuc:~/work/tinygo$ tinygo flash -monitor -target pico -size short -port /dev/ttyACM1 src/examples/echo/echo.go 
   code    data     bss |   flash     ram
   8548     108    3288 |    8656    3396
error: unable to search for a default USB device - use -port flag, available ports are /dev/ttyACM0, /dev/ttyACM1, /dev/ttyACM2

sfeldma@nuc:~/work/tinygo$ tinygo flash -monitor -target pico -size short src/examples/echo/echo.go 
   code    data     bss |   flash     ram
   8548     108    3288 |    8656    3396
error: failed to flash /tmp/tinygo1733382844/main.uf2: unable to locate device: RPI-RP2
```

I'm on dev branch of tinygo:

```
sfeldma@nuc:~/work/tinygo$ tinygo version
tinygo version 0.28.0-dev-2c0f61ca linux/amd64 (using go version go1.19.3 and LLVM version 15.0.0)

```

bxparks, 2023-04-06:
It gets weird even if you have 2 different boards connected: for example one with native USB (e.g. XIAO on /dev/ttyACM0), and one with a serial-USB converter (e.g. Arduino Nano on /dev/ttyUSB0). The `tinygo flash` seems to be able to figure it out, but `tinygo monitor` gets confused when it sees 2 active serial ports.

I make a side comment about this at the end of the "Multiple Microcontroller" subsection in the tutorial that I wrote (this is merged, but not deployed yet):
https://deploy-preview-329--tinygo.netlify.app/docs/tutorials/serialmonitor/#multiple-microcontrollers

If you try to provide the `-ports` flag, I recall that this info is used by `tinygo flash`, but not by the `tinygo monitor` subcommand invoked by the `-monitor` flag. 

I guess if this problem was important enough for me, I'd dive in and try to figure out the code to fix this. But in practice, it's been far easier to just disconnect all the other microcontrollers except for the one that I'm interested in. My USB hubs have an on/off switch on each port.

deadprogram, 2023-04-25:
I think the main problem in this issue has been addressed by #3640 so marking to close on next release.

deadprogram, 2023-04-25:
Had to revert, it was not working as expected.

davecheney, 2026-10-27:
Sorry for the delay. I tested this with TinyGo 0.44-dev (`b7ec945f`) on macOS arm64, with a Pico 2 and Pico 2 W connected simultaneously. Each board ran a program printing a distinct serial label.

Standalone `tinygo monitor -port=…` selected the expected board on all six connections. Combined `tinygo flash -monitor -port=…` also selected the expected board on all four checks, two per board.

Could you retry with 0.44-dev on Linux and let us know whether it still selects the wrong board?

PR #3640 (2023-04-25): Improve handling of multiple identical MCUs connected via USB
At least partially fixes #3637 

There are a few things:

1. Don't silently fail to flash if there is an error doing reset using the baud-rate trick because no port could be found

2. Improve error message when there are multiple candidate ports with same VID/PID by fixing the conditions when different errors are shown

3. Pass the port to the Monitor function after flashing (not this already works when using the `monitor` command)