tinygo-org/tinygo #3639: tinygo flash does not work on Seeed XIAO M0 on latest UF2 bootloaders

bxparks, 2023-04-06:
(This ticket captures the information that I posted in a Slack thread regarding flashing problems with the Seeed XIAO M0.)

As recently as Dec 2022, the Seeed Studio XIAO M0 dev boards shipped with a bootloader from Nov 2019: `UF2 Bootloader v3.7.0-33-g90ff611-dirty SFHWRO`. This appears to be a fork from [Adafruit's UF2](https://github.com/adafruit/uf2-samdx1) release [v3.7.0](https://github.com/adafruit/uf2-samdx1/releases/tag/v3.7.0) in Jun 2019, with the addition of a [small commit to support the XIAO](https://github.com/Seeed-Studio/uf2-samdx1/commit/90ff61102374b5432db9a724e1b496ce048b6667).

This default bootloader on the XIAO fails to flash about 25% time of using `tinygo flash`. When it stops flashing, power cycling off and on does *not* fix the problem. The Mass Storage Drive (MSD) refuses to appear. The solution is to hit the reset twice, like a double-click. But this process requires a tweezer or a small wire for the XIAO (see photo below).

Adafruit added support for the XIAO M0 in [v3.11.0](https://github.com/adafruit/uf2-samdx1/releases/tag/3.11.0) on Oct 27, 2020 containing this [PR#133](https://github.com/adafruit/uf2-samdx1/pull/133). The latest Adafruit UF2 is [v3.15.0](https://github.com/adafruit/uf2-samdx1/releases/tag/v3.15.0). It was hoped that the newer bootloaders would have a lower failure rate during flashing. Unfortunately, neither v3.11.0 nor v3.15.0 is compatible with `tinygo flash`. The failure rate is 100%. The error message is:
```
$ make flashxiao
tinygo flash -x -target=xiao
ld.lld --emit-relocs --gc-sections -L /usr/local/lib/tinygo -T targets/atsamd21.ld -o /tmp/tinygo2446135597/main /tmp/tinygo2446135597/main.o /usr/local/lib/tinygo/pkg/thumbv6m-unknown-unknown-eabi-cortex-m0plus/compiler-rt/lib.a /home/brian/.cache/tinygo/obj-d68421b1be1ecfe84508f94cbccad53404ece0565ee584e65f71a39c.bc /home/brian/.cache/tinygo/obj-37fa4132c4d26e97fcb584ca4a50c2229dc6be6c500914b575f3b29f.bc /home/brian/.cache/tinygo/obj-22541216a8f7c2ed3db6a1757d18d1a9870ced4fdcd0be2d5b0bde9f.bc /home/brian/.cache/tinygo/obj-bc9bc522b5d877fc580ef796adce33b99109eb29db6e31a2704cb6ea.bc /usr/local/lib/tinygo/pkg/thumbv6m-unknown-unknown-eabi-cortex-m0plus/picolibc/lib.a -mllvm -mcpu=cortex-m0plus --lto-O2 --thinlto-cache-dir=/home/brian/.cache/tinygo/thinlto -mllvm --rotation-max-header-size=0
error: failed to flash /tmp/tinygo2446135597/main.uf2: unable to locate device: Arduino
```

With the newer UF2 bootloaders, the MSD of the UF2 almost always appears on the host machine. Manually drag-and-droping the `/tmp/tinygoXXX/main.uf2` file into the MSD works perfectly. But `tinygo flash` does *not* work.

**Environment**:
* TinyGo Version: tinygo version 0.27.0 linux/amd64 (using go version go1.20.2 and LLVM version 15.0.0)
* OS: Linux Mint 21 (Ubuntu 22.04)

**Arduino IDE 1.8.19 Comparison**:
* The latest [Seeeduino Core 1.8.3](https://wiki.seeedstudio.com/Seeed_Arduino_Boards/) uses `bossac` v1.7 to perform its flashing. It works fine with the latest Adafruit UF2 bootloader v3.15, and with the older v3.11 bootloader.

**Reset Procedure**:
The reset line on the XIAO M0 dev board is exposed as a conductive pad on the PCB board:

[image]

kenbell, 2023-04-06:
Can you take a look in targets/xiao.json in tinygo - it should look like this:
```json
{
    "inherits": ["atsamd21g18a"],
    "build-tags": ["xiao"],
    "serial": "usb",
    "serial-port": ["2886:802f"],
    "flash-1200-bps-reset": "true",
    "flash-method": "msd",
    "msd-volume-name": "Arduino",
    "msd-firmware-name": "firmware.uf2"
}
```

If the device is resetting so that the volume appears, then you might need to update the "msd-volume-name" field to match the name of the volume you see on your system.

If the device isn't resetting at all, you might need to check the USB device id of the serial port for the device and change the "serial-port" field.

bxparks, 2023-04-06:
Wow, that's exactly it.
The original Seeed bootloader shows up as "Arduino".
The new bootloader from Adafruit shows up as "Seeed XIAO".
When I change the "msd-volume-name" to that, `tinygo flash` works.

I guess as long as Seeed Studio ships the old bootloader, we have to keep using "Arduino" as the volume name.

I will maintain an extra `targets/xiao-new.json` file locally, for my XIAOs with the new UF2 bootloader.

bxparks, 2023-04-06:
In case anyone else runs into this problem, here are the instructions for recompiling the original bootloader (or something very close to it) from the source code. This assumes an Ubuntu Linux machine:

1. Clone the Seeed Studio's fork of UF2:
    * `$ git clone git@github.com:Seeed-Studio/uf2-samdx1.git`
3. Check out the commit where the original XIAO bootloader was created:
     * `$ git checkout 90ff611`
4. You must obtain the `arm-none-eabi-gcc` compiler to compile this source.
    * You cannot use a modern version of arm-none-eabi-gcc, because the more recent compiler will terminate with warnings as errors.
    * You must use an older version. Fortunately, the Seeeduino Arduino Core ships with the ARM compiler.
    * Go to the Arduino IDE, install Seeeduino SAMD Board 1.8.3.
    * Locate the gcc directory at `$HOME/.arduino15/packages/Seeeduino/tools/arm-none-eabi-gcc/7-2017q4/bin`.
    * Add this to your `$PATH` variable.
5. Go into the git repo of the UF2 firmware
     * `$ cd uf2-samdx1`
6. Compile
    * `$ make BOARD=XIAO_m0`
7. Copy the resulting `.uf2` file to somewhere more accessible, like your `$HOME` directory.
    *  `$ cp build/XIAO_m0/update-bootloader-XIAO_m0-v1.23.0-263-g90ff611.uf2 $HOME`
8. In the file browser, drag-and-drop the `XIAO_m0-v1.23.0-263-g90ff611.uf2` into the UF2 disk drive of the XIAO. (You may need to use a tweezer to double-reset the board.)

That should revert the bootloader to the old version.

kenbell, 2023-04-16:
#3654 should fix / help with this.  `msd-volume-name` can then be an array and will match either "Arduino" or "Seeed XIAO".

aykevl, 2023-05-01:
Is this fixed with #3654?

bxparks, 2023-05-01:
Yes, I verified that this is fixed by #3654.
Closing.

closed 2023-05-01 by bxparks