tinygo-org/tinygo #3647: nil pointer dereference on wasi channel receive

nrwiersma, 2023-04-10:
Compiling the following application to WASI:
```
package main

func main() {}

var stopCh = make(chan struct{})

//export run
func Run() {
	<-stopCh
}

//export stop
func Stop() {
	close(stopCh)
}
```
with `tinygo build -o test.wasm -target=wasi .` using tinygo on my Mac M1 with version:
```
tinygo version 0.27.0 darwin/amd64 (using go version go1.20.3 and LLVM version 15.0.0)
```

and running with `wazero` returns:

```
panic: runtime error: nil pointer dereference
2023/04/10 20:07:17 wasm error: unreachable
wasm stack trace:
        .runtime.runtimePanic(i32,i32)
                0x377: /opt/homebrew/Cellar/tinygo/0.27.0/src/runtime/runtime_tinygowasm.go:72:6 (inlined)
                       /opt/homebrew/Cellar/tinygo/0.27.0/src/runtime/panic.go:59:7
        .runtime.nilPanic()
                0x163: /opt/homebrew/Cellar/tinygo/0.27.0/src/runtime/panic.go:122:14
        .run()
                0x3492: /opt/homebrew/Cellar/tinygo/0.27.0/src/runtime/chan.go:523:11 (inlined)
```

It is unclear to me if this is a `tinygo` issue or a `wazero` issue.

dgryski, 2023-04-10:
I believe this is effectively https://github.com/tinygo-org/tinygo/issues/2735 .

The runtime isn't being initialized when the functions are called.

nrwiersma, 2023-04-10:
I do know the `main` is being called as `_start`. What other initialisation is needed?

nrwiersma, 2023-04-10:
Ok, that was a dumb question. Once I move everything into the `main` and still use the channel to stop it, everything works as expected. Thanks for your help.

closed 2023-04-10 by nrwiersma