tinygo-org/tinygo #3650: USBHID not working in Digispark

BrunoTeixeira1996, 2023-04-12:
Hi guys,
I was playing with the USBHID feature in a Digispark device and I was following the https://github.com/tinygo-org/tinygo/pull/2889.
In my case looks like there's no `machine.BUTTON` and if I want to flash the device with the following code Ill get an error for undefined variables.

```go
package main

import (
	"machine"
	"machine/usb/hid/keyboard"
)

func main() {
	button := machine.BUTTON // UNDEFINED machine.BUTTON

	kb := keyboard.New()
	for {
		if button.Get() {
			kb.Write([]byte("tinygo"))
		}
	}
}
```

Flashing

```console
$ tinygo flash -target=digispark main.go 
# machine/usb/hid
/usr/local/lib/tinygo/src/machine/usb/hid/hid.go:35:11: undefined: machine.EnableHID
/usr/local/lib/tinygo/src/machine/usb/hid/hid.go:56:11: undefined: machine.SendZlp
/usr/local/lib/tinygo/src/machine/usb/hid/hid.go:64:10: undefined: machine.SendUSBInPacket
```

Any idea how I can solve this?
It would be cool to have a guide on how to setup the USBHID device btw

BrunoTeixeira1996, 2023-04-13:
I used the same code but for a XIAO microcontroler and the error does not appear anymore.
I guess this means the digispark support is not yet finished.
Is there a way I could help coding this support? any guide or something?

BrunoTeixeira1996, 2023-04-15:
https://github.com/digistump/DigistumpArduino/releases/download/1.6.7/digistump-avr-1.6.7.zip

This might help. In arduino they are using the `DigiKeyboard.h` file

BrunoTeixeira1996, 2023-04-17:
@deadprogram Thanks for the enhancement, I am available to help in this if needed.
I never did something similar but I can use my device to test code or even code anything in case it's needed.

aykevl, 2023-05-03:
While it is technically possible, I am not sure our current USB stack is lean enough for the Digispark, which _does not have USB hardware_ (so it needs to be bit-banged) and only has 512 bytes of RAM and 8kB of flash. It would still be cool to have USB support, though.

BrunoTeixeira1996, 2023-05-03:
@aykevl do you think its something plan for the future?

aykevl, 2023-05-03:
I'm not going to add support for this, but maybe someone else will?

BrunoTeixeira1996, 2023-05-03:
Ok sure, Ill wait since I have no clue on how I can help developing this :/