tinygo-org/tinygo #3652: Raspberry Pico - Change system clock

8lall0, 2023-04-13:
Since via the SDK is possible to set the clock of the board, i was wondering if tinygo offers this opportunity.

Thanks!

deadprogram, 2023-04-14:
Some initial work on this happening here https://github.com/tinygo-org/tinygo/pull/3398

8lall0, 2023-04-14:
Thanks @deadprogram , but i was talking about the "system" clock aka set the frequency of the chip. Probably i wrote my issue in the wrong way :')

aykevl, 2023-04-16:
We already have `CPUFrequency` and it's a function (that usually returns a constant) so that changing the CPU could be added. So I would agree with a `SetCPUFrequency`.

I am not aware of a way to do this at the moment, but it could be a useful addition.

Gustavomurta, 2023-10-11:
I'm an HW expert, not a SW expert. I realize that all the programming is already here to change the CPU clock frequency (RP2040). But I can't access these functions. Can anyone tell me how? I'm sorry for the lack of knowledge.

https://github.com/tinygo-org/tinygo/blob/release/src/machine/machine_rp2040_clocks.go

kenbell, 2023-10-13:
What's your use-case for modifying the clocks?  There's bunch of stuff that depends on PLL frequency that might break if the PLL frequency changes.

Gustavomurta, 2023-10-14:
My project is HW oriented. My goal is to change only the PLL Clock.

**Raspberry Pico - Tinygo my Experiments - PWM Digital Signal Generator**
https://forums.raspberrypi.com/viewtopic.php?t=357431

As the frequency is generated through CPU clock frequency dividers, the higher the frequency, the lower the precision (above 1 MHz). If the chosen frequency is proportional to the CPU clock, you can obtain very good accuracy.
In the future I will include programming to change the CPU clock, for better options.

kenbell, 2023-10-14:
None of those functions are public - changing the PLLs / clocks after tinygo is initialized is unknown.  It may break some peripherals, I guess?

I think you'll need to modify the code in `machine_rp2040_clocks.go`, and possibly `machine_rp2040_pll.go` in your local install of TinyGo to see what happens.  In Go only symbols starting with a capital letter are 'public'.

Gustavomurta, 2023-10-14:
> None of those functions are public - changing the PLLs / clocks after tinygo is initialized is unknown. It may break some peripherals, I guess?

I guess no. 
"The clocks block provides independent clocks to on-chip and external components. From these
sources it uses multiple clock generators to provide the required clocks. This architecture allows the user flexibility to
start and stop clocks independently and to vary some clock frequencies whilst maintaining others at their optimum
frequencies."

https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf?#page=181

> I think you'll need to modify the code in `machine_rp2040_clocks.go`, and possibly `machine_rp2040_pll.go` in your local install of TinyGo to see what happens. In Go only symbols starting with a capital letter are 'public'.

Thanks for this information. I didn't know