tinygo-org/tinygo #3659: compile the source code got error undefined: run_build_sh

epowsal, 2023-04-16:
2023 4 16 master source code

error:
M:/work/tool/go1.20.3.windows-amd64/bin/go.exe build [M:/work/code/go/src/github.com/tinygo-org/tinygo]
# tinygo.org/x/go-llvm
..\..\..\tinygo.org\x\go-llvm\llvm_dep.go:18:7: undefined: run_build_sh

aykevl, 2023-04-19:
I suspect you are trying to install TinyGo using `go install` on Windows, is that correct?
This is not supported, see https://tinygo.org/docs/guides/build/#building-tinygo. You need to build LLVM and then build TinyGo using `make` as described in the build guide.

aykevl, 2023-05-01:
Closing, as I believe this is due to `go install` on Windows.
If that's not the issue, then please comment.

closed 2023-05-01 by aykevl