tinygo-org/tinygo #3667: net.Dialer , os.Rename undefined

cnn-rnn, 2023-04-20:
I am getting 

type net.Dialer has no field or method Dial

and 

undefined: os.Rename

when trying 

tinygo build -target wasm 

What does this mean, and how to fix it?

cnn-rnn, 2023-04-20:
OK, I see that it is not implemented as per
https://github.com/tinygo-org/tinygo/blob/release/src/net/dial.go

What is the best way to implement it, for a specific project?

davecheney, 2026-10-24:
Both reported APIs are now declared. This program compiles with `tinygo build -target=wasm -o check.wasm main.go`:

```go
package main

import (
    "net"
    "os"
)

func unusedDial(d *net.Dialer, address string) (net.Conn, error) {
    return d.Dial("tcp", address)
}

func unusedRename(oldpath, newpath string) error {
    return os.Rename(oldpath, newpath)
}

func main() {
    println("Dialer.Dial and os.Rename compile; neither is called")
}
```

Checked with TinyGo 0.44.0-dev-eae3f875 and Go 1.27.3. This checks API availability only. It does not demonstrate working TCP connections or file renaming in a browser; runtime support depends on the target and host environment.

Closing the original missing-symbol report. Please open a new issue with a reproducer if you encounter further problems.

closed 2026-10-24 by davecheney