tinygo-org/tinygo #3668: Desktop operating system target support

programmingkidx, 2023-04-21:
Tinygo is made to produce small binaries which is what I want. A simple "Hello World" program should not be over 100k in size. This compiler can make some impressively small programs. I ask that desktop operating system targets be added to the target list. Mac OS and Windows users want programs that are small in size too.

aykevl, 2023-04-21:
All these systems are supported, but the [documentation](https://tinygo.org/docs/guides/linux/) is way out of date. The following are supported:
* linux/amd64
* linux/386
* linux/arm64
* linux/arm
* darwin/amd64
* darwin/arm64
* windows/amd64
* windows/arm64

`tinygo build` without `-target` will build for the current OS/arch, and the GOOS/GOARCH environment variables work as you'd expect. There is no need for any cross compiler packages, and CGo works out of the box without any setup.

programmingkidx, 2023-04-21:
I followed the directions from here: https://tinygo.org/getting-started/install/macos/, then made a simple "hello world" program. The result was a program that was only 76k in size😳. This is awesome!!!

So who do we contact to update the documentation?

aykevl, 2023-04-29:
The documentation was updated, see: https://github.com/tinygo-org/tinygo-site/pull/350

programmingkidx, 2023-05-01:
Could we add information about Windows and Mac OS support on this page https://github.com/tinygo-org/tinygo? 
It would be very useful under the "Supported boards/targets" section.

deadprogram, 2023-05-01:
@programmingkidx this infor was added to the `dev` branch here:

https://github.com/tinygo-org/tinygo/commit/36bd66a858e1c5ef33cd9da57eecacbbdb0d9ef6#diff-b335630551682c19a781afebcf4d07bf978fb1f8ac04c6bf87428ed5106870f5R85

Also to the website `dev` branch here: 
https://github.com/tinygo-org/tinygo-site/pull/350

Tagging this issue to be closed on next release. Thanks!

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing. Thank you everyone!

closed 2023-06-14 by deadprogram