tinygo-org/tinygo #3671: rp2040: Transaction with a NDIR CO2 sensor completely disables I2C functionality

soypat, 2023-04-23:
A specially crafted transaction will consistently crash the I2C peripheral. This happens after the second `Tx` call when configuring the [Sandbox Electronics 100,000ppm MH-Z16 NDIR CO2 Sensor](https://sandboxelectronics.com/?product=100000ppm-mh-z16-ndir-co2-sensor-with-i2cuart-5v3-3v-interface-for-arduinoraspeberry-pi). 

This driver works fine on the Raspberry Pi 3 and 4 (using Go version) and the ESP32 (using the C++ version), so I suspect it is a RP2040 issue.I have replicated this issue on a Raspberry Pi Pico and a Waveshare Zero. 

I have not been able to reproduce this crash trying to pull SDA permanently high or low. A Pull-up resistor to smooth out the SDA line did not seem to work either.

I rolled back the changes made in [#2743](https://github.com/tinygo-org/tinygo/pull/2743) and this still happens.

## Steps to reproduce
Run the example program here https://github.com/tinygo-org/drivers/pull/551/files

## Events
1. Abort (code number 8) during wait for stop condition: [`!i2c.interrupted(rp.I2C0_IC_RAW_INTR_STAT_STOP_DET)`](https://github.com/tinygo-org/tinygo/blob/0d56dee00f49bd50eb373c02c30062a75ec28f10/src/machine/machine_rp2040_i2c.go#L306)
2. Then it hits the timeout condition in that for loop
3. Returns a `errI2CWriteTimeout`
4. SCL line stays low after this "crash"
5. Future I2C transactions fail returning `ErrRP2040I2CDisable` during [I2C.disable()](https://github.com/tinygo-org/tinygo/blob/0d56dee00f49bd50eb373c02c30062a75ec28f10/src/machine/machine_rp2040_i2c.go#L251)

After the first event no other I2C transaction succeeds and the SCL line stays low. 

<details><summary>Screenshot of I2C transactions</summary>

[image]
[image]

</details>

soypat, 2023-04-28:
Dumping possibly related information from a quick glance at the datasheet and source code.

* The `restartOnNext` field of the I2C type is apparently unused and always false.

#### 4.3.7.1: Tx FIFO Management
> The component does not generate a STOP if the Tx FIFO becomes empty; in this situation the component holds the SCL
line low, stalling the bus until a new entry is available in the Tx FIFO. A STOP condition is generated only when the user
specifically requests it by setting bit nine (Stop bit) of the command written to IC_DATA_CMD register. 

#### Note 4.3.10.1.2. Slave-Transmitter Operation for a Single Byte
> Because the DW_apb_i2c’s Tx FIFO is forced into a flushed/reset state whenever a TX_ABRT event occurs, it is
necessary for software to release the DW_apb_i2c from this state by reading the IC_CLR_TX_ABRT register before
attempting to write into the Tx FIFO. See register IC_RAW_INTR_STAT for more details

#### Note 4.3.10.1.3. Slave-Receiver Operation for a Single Byte
> If the Rx FIFO is completely filled with data when a byte is pushed, then the DW_apb_i2c slave holds the I2C SCL line
low until the Rx FIFO has some space, and then continues with the next read request.

#### 4.3.10.3. Disabling DW_apb_i2c
> The DW_apb_i2c Master can be disabled only if the current command being processed—when the ic_enable deassertion occurs—has the STOP bit set to one. When an attempt is made to disable the DW_apb_i2c Master while
processing a command without the STOP bit set, the DW_apb_i2c Master continues to remain active, holding the SCL
line low until a new command is received in the Tx FIFO. When the DW_apb_i2c Master is processing a command
without the STOP bit set, you can issue the ABORT (IC_ENABLE.ABORT) to relinquish the I2C bus and then disable
DW_apb_i2c

soypat, 2023-06-04:
Below is a healthy I2C transaction during initialization (first two write registers) on the raspberry pi 4 using its hardware I2C along with the Saleae transactions from calling Init, CalibrateZero, and a couple Updates with values 405, 406 ppm read . Channel 0 is SDA, channel 1 is SCL. 
[image]
**Pi 4 Healthy transactions**: [ndir_rpi4.zip](https://github.com/tinygo-org/tinygo/files/11644625/ndir_rpi4.zip)
**Pico faulty transaction**: [ndir_pico_crash.zip](https://github.com/tinygo-org/tinygo/files/11644631/ndir_pico_crash.zip)

Still investigating...

soypat, 2023-06-04:
Apparently the issue was that an abort could be in fact encountered while waiting for a stop condition, [something the original C code was worried about](https://github.com/raspberrypi/pico-sdk/blob/f396d05f8252d4670d4ea05c8b7ac938ef0cd381/src/rp2_common/hardware_i2c/i2c.c#L191-L193). The way I got around to fixing this was adding a abort check between the write and read sections of the `tx` method. Will produce a PR after cleaning up.

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing. Thank you everyone!

PR #3770 (2023-06-04): RP2040 i2c: fix crash when encountering abort while waiting for stop condition
Resolves #3671, see issue, it's got all of the context. This PR now allows users to use the NDIR CO2 sensor and other peripherals that could fail during last bits of a write.

**Main Feature:** Fix unrecoverable I2C crash by clearing abort after waiting for stop condition.

**Additional features**:
- Removed the unused `i2c.restartOnNext` and set RESTART bit condition on the first byte being transmitted in a transaction, whether it be a write or read op.
- Added `Reasons` method users can access via an interface cas [... cut]
closed 2023-06-14 by deadprogram