tinygo-org/tinygo #3672: Decision: Ship with allocation panic in interrupt handlers

kenbell, 2023-04-24:
The current dev branch panics on any attempt to perform allocation inside an interrupt handler.

Currently, there is a known issue with the Bluetooth stack performing such allocations.  There may be other cases not yet discovered. 
 We need to decide how to handle allocations in interrupt handlers in v0.28.0, either:

1. Disable the panic by default [make it opt-in]
2. Enable the panic by default, but allow it to be disabled (using build tags?) [make it opt-out]
3. Keep the panic as mandatory (no way to disable)

dkegel-fastly, 2023-04-24:
The route from "bad feature present" to "bad feature absent" usually passes through "bad feature deprecated and off by default"... worth the effort if there is a community that would rise up with pitchforks otherwise.

dkegel-fastly, 2023-04-25:
oooh, the new option to show better panics may help here.

aykevl, 2023-04-25:
It has always been "deprecated" in some sense, in that the documentation said "don't do that". But of course, people still do (especially as long as the code appears to work just fine and the bug usually isn't obvious from the source code).

Making this opt-in probably doesn't help much. Defaults tend to be popular, and as long as code keeps working I don't think many people will change their code.
I'd be fine with making this opt-out (using a build tag or something) and removing the opt-out in the next release. That gives people time to fix their code, but gives them a strong incentive to fix the code.

sago35, 2023-04-25:
I think the biggest problem is that people don't know why they crash when they upgrade their tinygo version.
At the very least, I think the message during panic needs to be improved.

aykevl, 2023-04-25:
I've updated the message here: https://github.com/tinygo-org/tinygo/pull/3676

aykevl, 2023-05-04:
I think all the major blockers to enable this by default are fixed.

Any objections to enabling panics by default in the next release?
For those with objections: what criteria should be met before enabling this feature?

dkegel-fastly, 2023-05-09:
I think that means this issue can be closed as completed.

deadprogram, 2023-05-14:
Any other comments @kenbell @sago35 anyone else before I close this?

kenbell, 2023-05-15:
I think we're good.

deadprogram, 2023-05-16:
Thank you everyone, now closing.

closed 2023-05-16 by deadprogram