tinygo-org/tinygo #3683: Use two bits of type meta byte for Comparable / IsBinary

dgryski, 2023-04-27:
These two queries still require looping over all the fields in a struct, recursively.  Caching them in the meta byte would make both these operations `O(1)`.

deadprogram, 2023-05-16:
This is the only open issue for milestone 0.28 @dgryski :cat2:

deadprogram, 2023-05-17:
How about this one @dgryski ?

dgryski, 2023-06-02:
So we only *need* these flags for structs -- everything else it's just an easy type switch.  I suppose we could either choose to use these flags instead of the type switch *or* we could add a flag byte to the struct type that includes not only these two flags but also the alignment for the struct (so we don't need to recursively calculate that either).

deadprogram, 2023-06-05:
Tagging @aykevl to answer this question, please.

aykevl, 2023-06-06:
No strong preference from me.

> So we only _need_ these flags for structs -- everything else it's just an easy type switch.

True. Although, these bits are currently unused so we might as well generalize to all the types - not just structs (unless you know of a better use for these bits).

A third option would be to store the bits in the meta byte, but _only_ for the struct type. I don't see a real advantage of that though - except perhaps saving a few bits for future use in non-struct types.

dgryski, 2023-06-06:
I suppose since these are internal flags, we can always change it in the future if it turns out we have a better use for those bits.

PR #3773 (2023-06-08): compiler,reflect: use two bits of the meta byte for comparable/isBinary
Fixes #3683
commit 37849c48 (2023-06-08): compiler,reflect: use two bits of the meta byte for comparable/isBinary

Fixes #3683
closed 2023-06-10 by aykevl