tinygo-org/tinygo #3684: Use pointer tagging for `*T`, `**T`, and `***T`.

dgryski, 2023-04-27:
```
- Remove the ptrTo field, it won't be necessary anymore.
- Make sure all type structs are 32-bit aligned. This means the lower two bits of the pointer are zero.
- A type *T can then be represented by adding one to the pointer (this is equivalent to setting the lower bit to one).
- A type **T can be represented by adding two to the pointer.
- Same for ***T. A type ****T cannot be represented this way, we may still need a pointer type.
- Kind() returns PointerType when the lower two bits aren't zero, or else it behaves as it already does.
- Elem() subtracts one from the pointer if it is a pointer type, else it works as it already does.
```

deadprogram, 2023-05-16:
This is one of the few remaining open issues for the 0.28 milestone @dgryski :smile_cat:

dgryski, 2023-05-16:
The current partial fix for this is https://github.com/tinygo-org/tinygo/pull/3691 but I hit the limit of my current knowledge (and had some other things to work on).  Needs @aykevl to figure out how to solve the remaining issue with the optimization passes.

deadprogram, 2023-05-17:
Calling @aykevl we need you! :stethoscope:

deadprogram, 2023-05-22:
Now that https://github.com/tinygo-org/tinygo/pull/3737 has been merged, #3691 should be unblocked, @dgryski

deadprogram, 2023-06-06:
Moving this to release milestone 0.29 based on Slack conversation.

dgryski, 2023-06-06:
The tagged pointer pr should be good to merge.

deadprogram, 2023-06-06:
Does that mean we should merge it now?

dgryski, 2023-06-06:
It just needs a final review.

deadprogram, 2023-06-06:
Also which is the PR you are referring to?

deadprogram, 2023-06-06:
@dgryski were you referring to https://github.com/tinygo-org/tinygo/pull/3691 ?

dgryski, 2023-06-06:
Yes.

deadprogram, 2023-06-14:
This is part of the `v0.28` release so now closing. Thank you everyone!

closed 2023-06-14 by deadprogram