tinygo-org/tinygo #3685: wasm: allow GOOS=wasip1 to substitute for target=wasi

codefromthecrypt, 2023-04-27:
To allow drop-in replacement with go 1.21, it would seem nice to support `GOOS=wasip1` as an alternative to `-target=wasi`.

Per activity on https://github.com/golang/go/issues/58141, `GOOS=wasip1` will ship in Go 1.21 even if a subset of functionality vs the TinyGo equivalent target.

dgryski, 2023-04-27:
I agree we should do this. We also need to figure out how this will affect build tags. Do we migrate all of tinygo to the new tag to match upstream and add it to the list of tags exported to users, etc.

aykevl, 2023-04-27:
Same, we should add support for `GOOS=wasip1`. And perhaps also `-target=wasip1` for consistency.

deadprogram, 2023-08-10:
Is anyone working no this one yet?

aykevl, 2023-08-10:
I've made a draft PR for this feature: https://github.com/tinygo-org/tinygo/pull/3861

deadprogram, 2023-08-19:
Now that #3861 has merged, I am marking this PR to close on next release. Thanks everyone!

deadprogram, 2023-08-26:
Closing since this is part of release 0.29.0 thank you very much everyone!

closed 2023-08-26 by deadprogram