tinygo-org/tinygo #3701: Add support for PowerPC Darwin target

programmingkidx, 2023-05-02:
Hi, I ask that a new target be added to TinyGo that supports PowerPC Darwin. It would be very interesting to see Go programs that can work on Mac OS 10.4. Thank you.

deadprogram, 2023-05-02:
@programmingkidx that would be pretty difficult as the LLVM support we would need for this was removed a while ago.

https://discourse.llvm.org/t/want-to-add-powerpc-apple-darwin-as-target-what-is-involved/4536

aykevl, 2023-05-02:
Hmm, yeah this doesn't seem like something we can support. At the very least we would need support in LLVM. And I expect it will be difficult even with LLVM support because this hardware is so old.

Closing for now, as I don't think there is much we can do from the TinyGo side to support Darwin/PowerPC. Sorry about that.

closed 2023-05-02 by aykevl