tinygo-org/tinygo #3708: FlashBuffer.Write hangs if more than 256 bytes are attempted to be written

sago35, 2023-05-05:
It only worked correctly with 256 bytes as follows.
Tested with macropad-rp2040.

```go
package main

import (
	"fmt"
	"log"
	"machine"
	"time"
)

func main() {
	err := run()
	if err != nil {
		log.Fatal(err)
	}
}

func run() error {
	time.Sleep(2 * time.Second)
	println("Flash data start:      ", machine.FlashDataStart())
	println("Flash data end:        ", machine.FlashDataEnd())
	println("Flash data size, bytes:", machine.Flash.Size())
	println("Flash write block size:", machine.Flash.WriteBlockSize())
	println("Flash erase block size:", machine.Flash.EraseBlockSize())
	println()

	//addr := machine.FlashDataStart()
	flash := machine.OpenFlashBuffer(machine.Flash, machine.FlashDataStart())
	const offset = 0x00100000
	buf := [256]byte{}
	sz := len(buf)

	if true {
		for i := range buf {
			buf[i] = byte(i)
		}
		flash.Seek(offset, 0)
		_, err := flash.Write(buf[:])
		if err != nil {
			return err
		}
	}

	addr := machine.FlashDataStart() + offset
	flash.Seek(offset, 0)
	for {
		_, err := flash.Read(buf[:])
		if err != nil {
			return err
		}
		fmt.Printf("%08X % X\n", addr, buf)
		addr += uintptr(sz)
		time.Sleep(100 * time.Millisecond)
	}

	return nil
}
```

deadprogram, 2023-05-05:
I think the problem here, is that as currently implemented the `FlashBuffer` `Write()` does not handle correctly the case where it is called multiple times. The reason is that it always tries to erase blocks. Please see

https://github.com/tinygo-org/tinygo/blob/dev/src/machine/flash.go#L108

A better implementation would be able to determine if a block erase is needed or not, and handle this accordingly.

Code that uses the `BlockDevice` directly does not have this problem if it keeps track of this internally. This is just a shortcoming of the current `FlashBuffer` implementation.

aykevl, 2023-05-05:
> I think the problem here, is that as currently implemented the `FlashBuffer` `Write()` does not handle correctly the case where it is called multiple times. The reason is that it always tries to erase blocks.

This looks like a bug to me. A given flash area needs to be erased before calling `Write` (and the application should ensure that it is, because the machine package doesn't know what the correct behavior is in this case).

From this, I would actually suggest removing `Read` and `Write` as it is not possible to implement methods like this as if we're working on a regular file (even if they are convenient!), and only use the interface as defined in `BlockDevice`. People who would like to use `Read` and `Write` are probably better off using a real filesystem.

deadprogram, 2023-05-05:
> People who would like to use Read and Write are probably better off using a real filesystem.

https://github.com/tinygo-org/tinygo/issues/3460 needs to be addressed

aykevl, 2023-05-05:
> #3460 needs to be addressed

Certainly. But right now I'd argue it's better to use `WriteAt` instead and manage erasing flash manually.

deadprogram, 2023-05-06:
@sago35 the reason you found that 256 worked is that is actually the minimum page size for writes:

```c
#define FLASH_PAGE_SIZE (1u << 8)
```

https://github.com/raspberrypi/pico-sdk/blob/f396d05f8252d4670d4ea05c8b7ac938ef0cd381/src/rp2_common/hardware_flash/include/hardware/flash.h#LL42C1-L42C34

I will add a PR with a fix for that shortly.

sago35, 2023-05-06:
> This looks like a bug to me. A given flash area needs to be erased before calling Write (and the application should ensure that it is, because the machine package doesn't know what the correct behavior is in this case).

Basically, the following flow can be implemented.

1. when writing a range smaller than EraseBlockSize
    * Read the same size as EraseBlockSize, then Erase, replace it with the data you want to write, and Write. 2.
2. In case of writing a range equal to or larger than EraseBlockSize
    * Erase and then Write.

This is the same as is done in the sdcard example. I believe WriteAt and ReadAt should essentially work the same way.

https://github.com/tinygo-org/drivers/blob/820298bfe3e9f8192e6c7c1147b95d7eee02eb46/sdcard/sdcard.go#L534-L609

aykevl, 2023-05-07:
> This is the same as is done in the sdcard example. I believe WriteAt and ReadAt should essentially work the same way.

This doesn't match the flash interface, which does _not_ erase the area first:

```
// WriteAt satisfies the io.WriterAt interface and writes data to the device,
// one page at a time, starting at the provided address. This method assumes
// that the destination is already erased.
```

It can be important to keep it this way, for two reasons:

1. Erase+rewrite would fail when the power is lost at the wrong time.
2. Flash has a limited number of erase cycles. Therefore, I think it's better to make flash erase explicit to the user.

deadprogram, 2023-05-09:
With the current changes in the `dev` branch I have been successful at using the LittleFS file system with onboard flash.

Having a "proper" file system is certainly a lot better than a less robust solution. However, if all you want to do is store a struct with some data, more or less, bringing in an entire file system adds a lot more overhead. The same can be achieved using the pure `BlockDevice` interface with not too much extra work to explicitly erase before writing. I will try it out in a branch and submit a PR with that idea in mind.

deadprogram, 2023-05-14:
Now that #3718 has been merged this issue does not apply, so closing.

closed 2023-05-14 by deadprogram