tinygo-org/tinygo #3719: Failure to initialize variables

sago35, 2023-05-10:
Both of the following codes should return the same value, but CDCHID is not successfully initialized.

```go
package main

import (
	"fmt"
	"machine/usb/descriptor"
)

func main() {
	fmt.Printf("%#v\n", descriptor.CDCHID.HID[2][116:119])
	fmt.Printf("%#v\n", descriptor.HIDUsageMaximum(0x1FFF))
}
```

```
$ tinygo run ./_x/binary/
[]byte{0x2a, 0x0, 0x0}
[]byte{0x2a, 0xff, 0x1f}
```

deadprogram, 2024-12-02:
@sago35 was this issue resolved?

sago35, 2024-12-02:
I have checked, and it seems there are no issues now. @deadprogram 

```go
package main

import (
	"fmt"
	"machine/usb/descriptor"
)

func main() {
	for i, b := range descriptor.CDCHID.HID[2][:] {
		fmt.Printf("%03d %02X\n", i, b)
	}
	fmt.Printf("%#v\n", descriptor.CDCHID.HID[2][135:138])
	fmt.Printf("%#v\n", descriptor.HIDUsageMaximum(0x1FFF))
}
```

output:

```
...
[]byte{0x2a, 0xff, 0x1f}
[]byte{0x2a, 0xff, 0x1f}
```

closed 2024-12-02 by sago35