tinygo-org/tinygo #3740: Using goroutine on pca10040 hangs the board

m-kru, 2023-05-22:
I have a very simple program.
```go
package main

import (
	"machine"
	"time"
)

func blink_led1() {
	led := machine.LED1
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Low()
		time.Sleep(time.Millisecond * 500)

		led.High()
		time.Sleep(time.Millisecond * 500)
	}
}

func main() {
	 blink_led1()
}
```

It works fine. However, if I replace `blink_led1()` with `go blink_led1()` the board hangs.
I can see at least two potential reasons for such a state being. 
1. There is a bug in the runtime for the pca10040 board.
2. Goroutines are not supported for pca10040. However, in such a case, the compiler should return an error.  Currently `tinygo flash -target pca10040 main.go` returns with success.

deadprogram, 2023-05-22:
Your code is not quite correct in that you need to keep the `main` function running to prevent the program from exiting.

The following program does what you expect:

```go
package main

import (
	"machine"
	"time"
)

func blink_led1() {
	led := machine.LED1
	led.Configure(machine.PinConfig{Mode: machine.PinOutput})
	for {
		led.Low()
		time.Sleep(time.Millisecond * 500)

		led.High()
		time.Sleep(time.Millisecond * 500)
	}
}

func main() {
        go blink_led1()

        // yields the processor to the other running goroutines without exiting.
        for {
                time.Sleep(time.Second)
        }
}
```

Hope that helps!

m-kru, 2023-05-22:
Ok this works. However, I think we can do better.

First off all `time.Sleep(time.Second)` consumes resources.

Secondly, based on the target it is clear whether the main function can exit or no. For any MCU the `main` cannot exit. The compiler can take care of this automatically.

deadprogram, 2023-05-22:
> time.Sleep(time.Second) consumes resources.

That is not how the TinyGo scheduler works, exactly. There are a number of threads about this in the GH issues if you were to search.

> Secondly, based on the target it is clear whether the main function can exit or no. For any MCU the main cannot exit. The compiler can take care of this automatically.

Please feel free to submit a PR proposing a change to this.

aykevl, 2023-05-22:
> I guess the context is switched to this particular goroutine once every second?

Yes. If that's a problem (it probably isn't), you can sleep for longer periods of time.

In any case, in a real program you'd either keep the main goroutine busy with other stuff (probably a mainloop of some sorts), or would do the blinking in the main function thereby avoiding the return from the main goroutine.

> > We follow the Go specification, and the Go specification says that a program ends once the main function returns. So that is what happens here.
> 
> This is what the specification says. However, a program on MCU shall never end.

Well those two are in conflict, and we have chosen to follow the Go specification in that case. What happens is that it stops the MCU by stopping all active goroutines and sleeping forever.

I don't think there is anything that needs to be done here, so closing.

m-kru, 2023-05-22:
Maybe issue a warning or error?

aykevl, 2023-05-22:
The Go compiler doesn't issue an error when main returns either, so I don't see why we should do that in TinyGo (especially when it depends on the platform).
Emitting an error may also break some legitimate programs, so it would be a backwards incompatible change.