tinygo-org/tinygo #3744: Load the wrong go version

kerthcet, 2023-05-23:
I. Some Context:
1. I didn't set the `GOROOT` env
2. go version: `go version go1.20 darwin/amd64`
3. tinygo version: `tinygo version 0.27.0 darwin/amd64 (using go version go1.17.1 and LLVM version 15.0.0)`
4. The only go1.17.1 exists in my laptop locates at `/usr/local/Cellar/go/1.17.1`, but I usually use the specifically go version under `/usr/local/bin/go`

II. What happened:
tinygo build reports error: `error: requires go version 1.18 through 1.20, got go1.17`

III. What I expected:
No error and actually tinygo should use go1.20, rather than go1.17

IV. Others:
I test with the latest code with `goenv.GorootVersionString(goenv.Get("GOROOT"))`, I got the right version number of `go1.20`

kerthcet, 2023-05-23:
What I solved here is remove the `/usr/local/Cellar/go`, then it reports `error: cannot locate $GOROOT, please set it manually`, so I set the $GOROOT, everything works well.

Maybe if user doesn't set the `GOROOT`, we should report the warning in advance in case loading the unexpected location.

Just in case someone else met this problem again.
/close

closed 2023-05-23 by kerthcet

aykevl, 2023-05-23:
Can you check whether either `/usr/local/go` or `/usr/local/opt/go/libexec` exist? They may be broken links.

aykevl, 2023-05-24:
> Both of them do not exists.

Huh, weird.

In any case, #3745 should at least make the detection of the Go toolchain a bit more sensible.