tinygo-org/tinygo #3746: Add Support for BlueMicro Family of boards (nRF52)

jpconstantineau, 2023-05-24:
Add support for the following boards:
- [BlueMicro](https://www.tindie.com/products/jpconstantineau/bluemicro-nrf52832/) (nRF52832)
- [BlueMicro840](https://www.tindie.com/products/jpconstantineau/bluemicro840/) (nRF52840) (similar to nice!nano)
- [BlueMicro833](https://www.tindie.com/products/jpconstantineau/ebyte-e73-2g4m08s1e-breakout-bluemicro833-pcba/) (nrf52833) 

I am the creator of these boards and they have been available on Tindie for a while.  I would like to add them to TinyGO.  I would be willing to do the work to get them included but before going ahead, I would like to know if a PR would be accepted to be merged in.

Both the BlueMicro840 and BlueMicro833 are currently in [CircuitPython](https://circuitpython.org/downloads?manufacturers=JPConstantineau).  I also have a few keyboards on there too but as these are for a specific purpose, I thought it would be best to leave these out of TinyGo as they can be flashed with a generic rp2040 Pico config.

Please comment on how I should proceed and I'll take care of the work to prepare a PR for inclusion.

jpconstantineau, 2023-05-25:
I started looking into this.  Here are a few comments and some questions:
## nRF52840:
I far as I could find, all that needs to be done is
- create target file
- create a board.go file

For both cases, I can use the nicenano as  the two boards are very similar.
I noted a key diff in the json target files between the dev and current release branches where the "msd-volume-name" is in [] vs isn't...  I am assuming this is a change across all boards.  I'll test the files with the release I have on my PC and start looking into building and testing from the dev branch.

I'll prepare a PR for it once it's ready.

## nRF52833
This one will require a fair bit of work as none of the other 833 boards used UF2 bootloader and flashing method.  I therefore will need some guidance on getting it supported.  I suspect some USB code from the 840 will need to be added, as well as UF2/softdevice support.

## nRF52832
I havent' looked too much into this one yet but as the original feather nrf52832 isn't there I'll have to rely on the nrf52832 dev kit (PCA10040) with the jlink as the programmer instead of using serial.

jpconstantineau, 2023-05-26:
Support for the BlueMicro833 will depend on support of the PCA10010 (nRF52833 Dev Kit).

aykevl, 2023-05-26:
> Support for the BlueMicro833 will depend on support of the PCA10010 (nRF52833 Dev Kit).

Why? That seems like an unrelated board to me?

> ## nRF52833
> 
> This one will require a fair bit of work as none of the other 833 boards used UF2 bootloader and flashing method. I therefore will need some guidance on getting it supported. I suspect some USB code from the 840 will need to be added, as well as UF2/softdevice support.

I expect most/all code can be shared with the nef52840. I don't think you need to write anything special for this chip.

jpconstantineau, 2023-05-26:
Why the PCA10100?  I have it here and don't have any other reference 833 boards (other than my bluemicro833). The 833 dev kit is practically identical to the 840 dev kit and porting one over to the other should be relatively simple.  I also agree that most code should port over without writing anything special.  It's the inheritance chain that I need to understand/look at more carefully.  For example, why is there a "machine_nrf52840_usb.go" and no "machine_nrf52833_usb.go" when the 833 clearly has USB as well...  I suspect this is because the microbit USB port doesn't connect to the core chip itself but the DAPLink chip (both nRF52 chips).   It's safer to start with the reference dev kit from Nordic than another 3rd party board.

Once that is done, getting UF2 working on the dev kit will pave the way for other boards.  I have been looking at flashing the uf2 bootloader to the dev kit and have been having issues.  I'll need to look at it as well as this is a pre-requisite for circuitpython as well.

jpconstantineau, 2023-06-01:
> > For example, why is there a "machine_nrf52840_usb.go" and no "machine_nrf52833_usb.go" when the 833 clearly has USB as well...
> 
> Probably just because nobody has got it working. We might need to rename that file to machine_nrf528xx_usb.

Moving this to #3750...