tinygo-org/tinygo #3748: out of bounds memory access with regexp.Compile

raffis, 2023-05-25:
## Reproduce
```
tinygo build -o mx.wasm -scheduler=none -target=wasi x.go
wasmtime run mx.wasm
```

x.go:
```
package main

import "regexp"

func main() {
	_, err := regexp.Compile(`[0-9A-Za-z-]{0,53}`)
	if err != nil {
		panic(err)
	}
}
```

Changing the same regex above to ``[0-9A-Za-z-]{0,52}`` works and no out of bounds occurs.
Is this expected? I see that regexp/syntax tests fail in the supported packages. 

## Output
(export WASMTIME_BACKTRACE_DETAILS=1)
```
Error: failed to run main module `mx.wasm`

Caused by:
    0: failed to invoke command default
    1: error while executing at wasm backtrace:
           0: 0x1a00 - (*regexp/syntax.compiler).alt
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:201:20              - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:154:13
           1: 0x20f8 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:145:18
           2: 0x2021 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:137:27
           3: 0x2125 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:147:27
           4: 0x2021 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:137:27
           5: 0x2125 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:147:27
           6: 0x2021 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:137:27
           7: 0x2125 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:147:27
           8: 0x2021 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:137:27
           9: 0x2125 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:147:27
          10: 0x2021 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:137:27
          11: 0x2125 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:147:27
          12: 0x2021 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:137:27
          13: 0x2125 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:147:27
          14: 0x2021 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:137:27
          15: 0x2125 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:147:27
          16: 0x2021 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:137:27
          17: 0x2125 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:147:27
          18: 0x2021 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:137:27
[... 151 lines cut ...]
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:137:27
          95: 0x2125 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:147:27
          96: 0x2021 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:137:27
          97: 0x2125 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:147:27
          98: 0x2021 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:137:27
          99: 0x2125 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:147:27
         100: 0x2021 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:137:27
         101: 0x2125 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:147:27
         102: 0x2021 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:137:27
         103: 0x2125 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:147:27
         104: 0x2021 - (*regexp/syntax.compiler).compile
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:137:27
         105: 0xbc37 - _start
                           at /home/raffi/.gvm/gos/go1.20/src/regexp/syntax/compile.go:74:16
    2: wasm trap: out of bounds memory access
```

aykevl, 2023-05-25:
Looks like a stack overflow to me. Have you tried increasing the stack size using `-stack-size=32k` for example?

raffis, 2023-05-25:
> Looks like a stack overflow to me. Have you tried increasing the stack size using `-stack-size=32k` for example?

I did not but increasing the stack size does not solve it, also not a value higher than 32k.

dgryski, 2023-05-26:
This is a known stack overflow issue. If possible, switch to a code-generation based regexps package like libfsm or ragel.

dgryski, 2025-04-07:
@JounQin Increasing memory is the only workaround at the moment assuming it's a stack overflow due to a large regex compilation step. However on native platforms you should have *much* more memory than on wasm.  Can you show me the regex compilation step that's failing?  I can't find it your PR.

JounQin, 2025-04-07:
@dgryski I don't know which part is broken.

https://github.com/un-ts/prettier/pull/430

I'm using https://github.com/mvdan/sh package at https://github.com/un-ts/sh-syntax/blob/63e69dc8613fa5fb7de11e4ac49c91494e749fe8/processor/main.go#L7.

JounQin, 2025-04-07:
After using:

```jsonc
// target.json
{
  "inherits": [
    "wasm"
  ],
  "ldflags": [
    "--initial-memory=52428800",
    "--max-memory=1073741824",
    "-zstack-size=1048576"
  ]
}
```

I seems passing now.

davecheney, 2026-10-26:
The reported expressions now compile and match successfully with TinyGo 0.44.0-dev-eae3f875 and Wasmtime 50.0.0, using `-target=wasip1 -scheduler=none` with the default stack. Both repetition limits, 52 and 53, passed. They also passed with a 1 MiB WebAssembly stack.

```go
package main

import (
	"fmt"
	"os"
	"regexp"
)

func main() {
	if len(os.Args) != 2 || (os.Args[1] != "52" && os.Args[1] != "53") {
		fmt.Fprintln(os.Stderr, "usage: check.wasm 52|53")
		os.Exit(1)
	}
	pattern := `[0-9A-Za-z-]{0,` + os.Args[1] + `}`
	re, err := regexp.Compile(pattern)
	if err != nil {
		fmt.Fprintln(os.Stderr, err)
		os.Exit(1)
	}
	if got := re.FindString("TinyGo-123!"); got != "TinyGo-123" {
		fmt.Fprintf(os.Stderr, "unexpected match %q\n", got)
		os.Exit(1)
	}
	fmt.Println("PASS:", pattern)
}
```

```sh
tinygo build -target=wasip1 -scheduler=none -o check.wasm main.go
wasmtime run check.wasm 52
wasmtime run check.wasm 53
```

```text
PASS: [0-9A-Za-z-]{0,52}
PASS: [0-9A-Za-z-]{0,53}
```

Closing because the original reproducer no longer fails. This does not rule out stack exhaustion with larger expressions. Please open a new issue with a minimal reproducer if you encounter a failure with a current TinyGo version.

closed 2026-10-26 by davecheney