tinygo-org/tinygo #3750: Add support for nRF52833 DK (PCA10100)

jpconstantineau, 2023-05-26:
Nordic Semiconductors has a Dev Kit for the nRF52833 which is very similar to the nRF52840. 
This board is missing support from tinygo.

I have one of these and would like to help to bring support of this board.

I'll start the work to support this board based on what's implemented for the nrf52840 dev kit (PCA10056).

jpconstantineau, 2023-06-01:
Moved from #3749 

> > For example, why is there a "machine_nrf52840_usb.go" and no "machine_nrf52833_usb.go" when the 833 clearly has USB as well...
> 
> Probably just because nobody has got it working. We might need to rename that file to machine_nrf528xx_usb.

If I change the machine_nrf52840_usb to machine_nrf528xx_usb and add the build flag for the 833, won't that also change the size for the microbit v2?  

From what I see, the microbit v2 is the only 833 board implemented and unfortunately it is without usb. We might change the build flag to include a !microbit...  I suspect there would be a better pattern to implement here...    Is there a board somewhere that "removed" features inherited from the parent?

Additionally, I would like to port over the machine_nrf52840_usb_reset_uf2 and machine_nrf52840_enter_bootloader files to the 833 as well.  They use the same bootloader and magic codes; these won't apply to the microbit as it probably use a different bootloader.

I have both a PCA10056 (nrf52840) and PCA10100 (nrf52833).  The boards have the same features, with few minor exceptions that depends on the processor.  They both can get loaded with the UF2 bootloader (which includes the softdevice). I would like to get to the point where Tinygo code from one could be flashed to the other.  I was able to test that I can flash code to both (button2.go), however, I think it wiped the bootloader and softdevice; hence both needing some work to bring the adafruit nrf52840 bootloader relevant files to the PCAs.

I'll need some guidance on how to proceed...

Btw, are there examples of tinygo code that works with the softdevice?

aykevl, 2023-06-01:
> If I change the machine_nrf52840_usb to machine_nrf528xx_usb and add the build flag for the 833, won't that also change the size for the microbit v2?

I don't know, the way to find out is to measure it.

> From what I see, the microbit v2 is the only 833 board implemented and unfortunately it is without usb. We might change the build flag to include a !microbit... I suspect there would be a better [...]

Don't do this, that's very unintuitive.
I haven't looked at this code specifically, but as long as the USB isn't initialized it _should not_ result in a binary size increase.

> I would like to get to the point where Tinygo code from one could be flashed to the other. I was able to test that I can flash code to both (button2.go), however, I think it wiped the bootloader and [...]

I'm not sure what you mean, you want to flash the same binary to both boards? Or compile the same source files (unmodified) for both boards? The latter is typically supported, the former isn't (it might work if you get lucky).

> however, I think it wiped the bootloader and softdevice; hence both needing some work to bring the adafruit nrf52840 bootloader relevant files to the PCAs.

Are you sure? As long as USB support isn't added you will need some special way to enter the bootloader (usually by pressing the reset button twice within a second or so).

jpconstantineau, 2023-06-02:
> Are you sure? As long as USB support isn't added you will need some special way to enter the bootloader (usually by pressing the reset button twice within a second or so).

I just checked the linker script for the PCA10056 and it refers to the nrf52840, which ask to flash at memory 0x00000000 instead of 0x00000000+0x26000.  This essentially overwrites the bootloader.  Tinygo code runs fine to blink but with the bootloader gone, you do need a jlink (which is on board of the PCA10056) to reflash the chip.  

I'll first create a new target for the pca10056 called pca10056-s140v6.json.  V6 is the softdevice version that's standard with the Arduino_nRF52_Bootloader.

aykevl, 2023-06-02:
Normally I would expect a UF2 bootloader to prevent overwriting itself. Did you maybe flash using JLink? That would certainly overwrite the bootloader. (Using `gdb` instead of `flash` also switches to a debugger interface like JLink).

jpconstantineau, 2023-06-02:
The default pca10056 flash method uses the jlink.  However, the linker script does have the memory address to which to write the program (those memory addresses mentioned above).  Ultimately, that's encoded in the hex file.  I'll be making a uf2 target for it and that should resolve things.