tinygo-org/tinygo #3754: type *sync.Map has no field or method CompareAndDelete

NanoRed, 2023-05-27:
it seems like tinygo compiler has not implement this method of sync.Map yet
i will be grateful that tinygo will support this at a certain time<3

dkegel-fastly, 2023-05-27:
For upstream context, see https://github.com/golang/go/issues/51972

Know any open source packages that depend on these new methods yet?

NanoRed, 2023-05-28:
not yet. i found this from using this methods for my own project

dkegel-fastly, 2024-04-01:
FWIW, https://github.com/getkin/kin-openapi seems to use this:

https://github.com/getkin/kin-openapi/blob/78bb273e5892da3b0c8fc31857499449adfaba6c/openapi3/schema_pattern.go#L27
```
	var _ bool = compiledPatterns.CompareAndSwap(pattern, nil, cp)
```
https://github.com/getkin/kin-openapi/blob/78bb273e5892da3b0c8fc31857499449adfaba6c/openapi3/schema.go#L53
```
    compiledPatterns sync.Map
```