tinygo-org/tinygo #3756: Flashing Arduino Nano fails with "out of sync" errors at baudrate 57600

toxyl, 2023-05-29:
## Description
I was experimenting with PWM functionality on an Arduino Nano and encountered issues while uploading my program. Since then, I continuously received "out of sync" errors from avrdude. To troubleshoot the problem, I followed these steps:
- Reverted my code back to a state that doesn't use PWM and successfully flashed a new Nano without any issues.
- Reintroduced the function to initialize PWM (but not the code to use it) and flashed the new Nano again. However, it failed to flash, and avrdude threw "out of sync" errors upon subsequent attempts.
- To rule out code-related problems, I used the Arduino IDE and an Arduino UNO as ISP to burn a fresh bootloader to both Nanos. After that, I uploaded the Blink example to both Nanos from the Arduino IDE successfully.
- Despite the successful uploads using the Arduino IDE, flashing the Nanos with tinygo still failed with "out of sync" errors.
- To further investigate, I obtained a third Nano and flashed it with a blank program, which worked without any problems. However, attempting the same with the other Nanos still resulted in failures.
- After examining the exact commands executed during flashing, I noticed that the Arduino IDE used a baudrate of 115200, while tinygo used 57200.
- Adjusted my tinygo build command to `tinygo flash -target arduino-nano -port=COM5 -baudrate=115200 -x .` and retried the flashing, but it still failed.
- Upon inspecting the output, I found that tinygo used the following avrdude command: `avrdude -c arduino -p atmega328p -b 57600 -P COM5 -U 'flash:w:[PATH TO HEX]:i'`.
- Ran the same command with a baudrate of 115200, and the flashing process completed successfully on all three Nanos.

After examining the commands used by the Arduino IDE and tinygo, it was discovered that the Arduino IDE successfully flashed the Nanos at a baudrate of 115200. However, tinygo ignored the specified baudrate and used 57600 instead. Running the avrdude command directly with a baudrate of 115200 allowed successful flashing on all three Nanos.

## Expected Behavior
The Nanos should be flashable at a baudrate of 57600, as it was the default baudrate used by tinygo but mine now seem botched and are only flashable at 115200. Setting the baudrate using the tinygo flash command should be reflected in the `avrdude` command used.

## Environment
- Three Arduino Nanos
- Operating System: Microsoft Windows 11 Pro 10.0.22621 Build 22621
- tinygo version: `tinygo version 0.27.0 windows/amd64 (using go version go1.20.3 and LLVM version 15.0.0)`

aykevl, 2023-05-31:
I think you need `-target=arduino-nano-new` to select the newer Arduino Nano variants.

Note that `-baudrate` is just for the `monitor` subcommand:

```
  -baudrate int
        baudrate of serial monitor (default 115200)
```

toxyl, 2023-05-31:
Ah, I see @ baudrate.

I've tested `arduino-nano-new` and it does indeed work, thanks! Do you happen to have an idea why that changed? Before experimenting with PWM I was able to use `arduino-nano` but since then it doesn't work anymore. I tried it with yet another one of my Nanos and encountered the same behavior. Before using PWM I was able to flash it several times as `arduino-nano` but once I've uploaded something with PWM to it, it stopped working. I've tested it, that Nano now also needs to be handled as `arduino-nano-new`. PWM being the cause might be a red herring, though, because I believe earlier versions of my code were buggy.

closed 2023-05-31 by toxyl

aykevl, 2023-06-01:
It seems very unlikely to me that using PWM would cause this. The only thing that might have an effect is the size of the binary. 

Do you also need to pick the flash baud rate in the Arduino IDE? If not, I wonder what commands they use to that work on both the old and new bootloader?

toxyl, 2023-06-01:
I just checked and the Arduino IDE automatically selects the new processor for which it uses 115200. If I select the old processor it switches to 57600. Unfortunately I have no Nano left that I can still flash with `arduino-nano`, so I'm not sure what the Arduino IDE would have done with that.

aykevl, 2023-06-01:
Ah, so you do have to pick the specific board (old or new)?

aykevl, 2023-06-01:
> I have to select the port (in my case COM4) and then it'll detect that an Arduino Nano is connected to it and, I think, it automatically selects the processor as well.

Oh interesting.

> It changed once I added PWM (in TinyGo) and flashed with that.

Are you sure about that? As far as I know, flashing is done entirely through a separate USB-UART chip and the program on it shouldn't matter at all. What _would_ matter, is the bootloader (so the behavior might have changed after you updated the bootloader).

toxyl, 2023-06-01:
The odd thing is that I made a web interface that calls the flash command, so I'm pretty sure that I haven't changed anything besides the code. It just stopped working and spat out errors instead while I've used it a bunch of times before without issues. PWM might totally be a red herring, I find the idea intriguing that some part of memory is in some way corrupted, but to test that I first need to get a Nano back into the previous state. With a second Nano I observed the same behavior, flashing via the web interface worked until I tried adding PWM. But if it has to do with program size, then that might just have been the piece that pushed it over the edge.

Can I use TinyGo to rewrite the bootloader using an Arduino UNO as ISP? In the Arduino IDE I can do that, so that I burn bootloaders.

aykevl, 2023-06-02:
> I find the idea intriguing that some part of memory is in some way corrupted, [...]

That's possible, but would be surprising. If you find this to be the case, that would be a bug (TinyGo should not overwrite the bootloader, and the bootloader should not allow overwriting itself).

> Can I use TinyGo to rewrite the bootloader using an Arduino UNO as ISP? In the Arduino IDE I can do that, so that I can burn bootloaders.

I'm not sure how you would use TinyGo for this job. I recommend just using the Arduino IDE for this, which is certainly going to be better tested.

toxyl, 2023-06-15:
So I've made an application that generates test programs which it then builds with `tinygo` and finally flashes the nano with using `avrdude`. I have been able to create situations where the nano locks up until reconnected, but I never got it to switch type. I suppose I've hit a very specific set of circumstances that I wasn't able to recreate in thousands of runs with different settings. I'll make a new issue if I manage to reproduce it.