tinygo-org/tinygo #3758: WASM file detected as invalid because bulk memory support is not enabled

k33g, 2023-05-31:
Hello,
I build a wasm module with TinyGo like this:
```bash
tinygo build -o hello-orange.wasm \
    -scheduler=none \
    --no-debug \
    -target wasi ./main.go 
```

I would like to publish it to https://wapm.io/, but when I try this command `wapm publish`, I get this error message:

```bash
Error: WASM file "hello-orange.wasm" detected as invalid because bulk memory support is not enabled (at offset 3196)
```
I will create an issue on the wasmer project, but I wanted to know if I need to activate a specific flag when I build the module?

fgsch, 2023-05-31:
Which tinygo version are you using?

aykevl, 2023-05-31:
We use bulk memory instructions, and there is no (easy) way to disable them at the moment.

> because bulk memory support is not enabled

Perhaps bulk memory can be enabled somewhere?

k33g, 2023-05-31:
@aykevl Thank you. Apparently, it is a wapm bug, but they told me that was fixed in a previous version and closed the issue (https://github.com/wasmerio/wasmer/issues/3932) I hope they will reopen it.

closed 2023-05-31 by k33g