tinygo-org/tinygo #3762: Allowing for non US keyboard keycodes for USB HID, e.g. UK

andrewfstratton, 2023-06-02:
Currently the HID libraries only seem to allow for US keyboard keycodes.  Please could they allow for different keyboard layouts, e.g. UK?

e.g. Using tinygo on a Badger2040 to send '#' (hash) using USB HID will instead show on a PC as '£' (UK Pound) when Windows is set with UK keyboard.

Without this, there is no (simple) way to send the '@' character, which shows as '"', as well as some other characters...

sago35, 2023-06-02:
Here is a setting for Japanese (JIS) keyboards.
I think there are keyboards from different countries and regions, so we need to think of something.

https://github.com/sago35/tinygo-keyboard
https://github.com/sago35/tinygo-keyboard/blob/ea336a62da237ce1e1972d8fc5d999a584e267c3/keycodes/jp/keycodes.go

aykevl, 2023-06-02:
Oof, that's a bug.

To be honest, I don't think we should include keycode to language/region specific character mapping in the TinyGo standard library. This should be an external library that can be updated separately. There are just way too many keyboard layouts out there to fully support in a low level USB package.

Example: we have two different keyboard layouts here in The Netherlands: the ISO layout and the US layout (I prefer the US layout, most people seem to use the ISO layout - they are very similar). I really don't want TinyGo to need to support these kinds of details.