tinygo-org/tinygo #3768: I2S support for ESP32 or RP2040 ?

NV4RE, 2023-06-04:
Hello, I'm starting using `tinygo` for quite sometime its build and run very fast make parallel very easy, but I feel like it kindda lacking some feature compare to other framework, like I2S only supported very few boards but popular boards such as ESP32 and RP2040 did not have this feature yet.

I2S support would open new opportunities for new projects e.g. `internet radio player`, `spotify player`, `keyboards`.

Are there plans to add i2s support for other board ?

soypat, 2023-06-04:
The RP2040 does not have hardware I2S. It'd probably have to happen through PIO or a software I2S implementation.

NV4RE, 2023-06-05:
> The RP2040 does not have hardware I2S. It'd probably have to happen through PIO or a software I2S implementation.

Thank you for the advice. I'm searching for an example of adding a custom program to the `PIO` but can't find any up-to-date with `Tinygo`. It would be very grateful if you could provide an example for the API.

soypat, 2023-06-05:
I'm afraid PIO is still not a feature of release TinyGo. It is still in process: https://github.com/tinygo-org/tinygo/issues/1953. 

I2S on the pico has been implemented via PIO here: https://github.com/raspberrypi/pico-extras/tree/master/src/rp2_common/pico_audio_i2s, this is in C.

soypat, 2023-11-01:
I2S support discussion can be moved to the pio repository where there is already a minimal, untested PIO implementation in the works https://github.com/tinygo-org/pio/blob/main/rp2-pio/piolib/i2s.go

closed 2023-11-01 by soypat

AutomationD, 2024-02-13:
Hi, I wanted to check-in, what's the status of I2S on esp32? I'm a bit confused it doesn't work but seems like it's not implemented at all? Seems like Esp8266 has it 🤔 

If not, how hard is it to implement it?

AutomationD, 2024-02-13:
Hi, @soypat ! Thank you for responding. 
Now it makes more sense, so basically they are not implemented for esp32 device.

I'm super new to tinygo, I am doubtful I can get this all the way though, but I'll try to dig into it.

Thanks!