tinygo-org/tinygo #3771: Unable to get the Whistle compiler to run under TinyGo latest dev - probably reflection?

deadprogram, 2023-06-06:
Trying to get the Whistle compiler to run under TinyGo. Seems like a possible problem related to reflection?

Please see https://github.com/deosjr/whistle/issues/11

awmorgan, 2023-06-07:
I noticed an optimized build panics in macro.go line 240 but an unoptimized build (-opt=0) does a lot of recursion in runtime.alloc -> runtime.runGC ...

```
/Volumes/git/whistle (main)$ (cd ../tinygo && git log --oneline -1 )
4a240827 (HEAD -> dev, origin/dev) main: add -internal-nodwarf flag
/Volumes/git/whistle (main)$ tinygo version
tinygo version 0.28.0-dev darwin/amd64 (using go version go1.20.4 and LLVM version 15.0.7)
/Volumes/git/whistle (main)$  git log --oneline -1 
90a4546 (HEAD -> main, origin/main, origin/HEAD) github actions workflow added (#7)
/Volumes/git/whistle (main)$ go build -o ./build/whistle ./cmd/whistle/
/Volumes/git/whistle (main)$ tinygo build -o ./build/tinywhistle ./cmd/whistle/
/Volumes/git/whistle (main)$ dsymutil ./build/tinywhistle 
warning: (x86_64) /var/folders/rp/fpc8t1293dn1r6jycrqn0hlw0000gn/T/tinygo262314359/embed-9a85c752476eff2548335f990c152e9e-1096280106.o unable to open object file: No such file or directory
warning: (x86_64) /var/folders/rp/fpc8t1293dn1r6jycrqn0hlw0000gn/T/tinygo262314359/embed-20a76981467986b18ed72dc855f2413b-37407959.o unable to open object file: No such file or directory
/Volumes/git/whistle (main)$ ls -latrh build/
total 5704
drwxr-xr-x  15 arthurmorgan  admin   480B Jun  6 19:16 ..
-rwxr-xr-x   1 arthurmorgan  admin   2.4M Jun  6 19:20 whistle
-rwxr-xr-x   1 arthurmorgan  admin   436K Jun  6 19:20 tinywhistle
drwxr-xr-x   5 arthurmorgan  admin   160B Jun  6 19:21 .
drwxr-xr-x   3 arthurmorgan  admin    96B Jun  6 19:21 tinywhistle.dSYM
/Volumes/git/whistle (main)$  ./build/tinywhistle ./examples/hello.lisp 
42
/Volumes/git/whistle (main)$  ./build/tinywhistle ./examples/datalog.lisp 
panic: pattern assumed to be list
Abort trap: 6
/Volumes/git/whistle (main)$ lldb -- ./build/tinywhistle ./examples/datalog.lisp 
(lldb) target create "./build/tinywhistle"
Current executable set to '/Volumes/git/whistle/build/tinywhistle' (x86_64).
(lldb) settings set -- target.run-args  "./examples/datalog.lisp"
(lldb) r
Process 86445 launched: '/Volumes/git/whistle/build/tinywhistle' (x86_64)
panic: pattern assumed to be list
Process 86445 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff2090290e libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff2090290e <+10>: jae    0x7fff20902918            ; <+20>
    0x7fff20902910 <+12>: movq   %rax, %rdi
    0x7fff20902913 <+15>: jmp    0x7fff208fcab9            ; cerror_nocancel
    0x7fff20902918 <+20>: retq   
Target 0: (tinywhistle) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff2090290e libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff209315bd libsystem_pthread.dylib`pthread_kill + 263
    frame #2: 0x00007fff20886406 libsystem_c.dylib`abort + 125
    frame #3: 0x0000000100005c56 tinywhistle`runtime._panic(message=<unavailable>) at panic.go:52:7 [opt]
    frame #4: 0x000000010001a2a7 tinywhistle`github.com/deosjr/whistle/lisp.unifyWithEllipsis(p=<unavailable>, q=<unavailable>, s=0x00000001010bb770, depth=<unavailable>) at macro.go:240:8 [opt]
    frame #5: 0x00000001000191a3 tinywhistle`github.com/deosjr/whistle/lisp.syntaxRules$1 [inlined] github.com/deosjr/whistle/lisp.unify(q=<unavailable>, s=0x00000001010bb770) at macro.go:218:26 [opt]
    frame #6: 0x0000000100019137 tinywhistle`github.com/deosjr/whistle/lisp.syntaxRules$1(p=<unavailable>) at macro.go:74:13 [opt]
    frame #7: 0x000000010001c7c0 tinywhistle`github.com/deosjr/whistle/lisp.expandMacro(p=<unavailable>) at macro.go:44:11 [opt]
    frame #8: 0x000000010001b586 tinywhistle`(p=<unavailable>, env=<unavailable>, e=(typecode = 0x0000000100038208, value = 0x00000001010bb5d0)).evalEnv at env.go:70:25 [opt]
    frame #9: 0x000000010001c290 tinywhistle`(p=0x000000010101bd50, env=0x0000000101039a50, e=<unavailable>).evalEnv at env.go:130:30 [opt]
(lldb) f 4
tinywhistle was compiled with optimization - stepping may behave oddly; variables may not be available.
frame #4: 0x000000010001a2a7 tinywhistle`github.com/deosjr/whistle/lisp.unifyWithEllipsis(p=<unavailable>, q=<unavailable>, s=0x00000001010bb770, depth=<unavailable>) at macro.go:240:8 [opt]
   237                  return true
   238          }
   239          if !p.isList {
-> 240                  panic("pattern assumed to be list")
   241          }
   242          if !q.IsPair() {
   243                  return false
(lldb) p p
error: Couldn't materialize: couldn't get the value of variable p: variable not available
error: errored out in DoExecute, couldn't PrepareToExecuteJITExpression
(lldb) up
frame #5: 0x00000001000191a3 tinywhistle`github.com/deosjr/whistle/lisp.syntaxRules$1 [inlined] github.com/deosjr/whistle/lisp.unify(q=<unavailable>, s=0x00000001010bb770) at macro.go:218:26 [opt]
   215  // TODO: for now, all symbols are pattern variables
[... 129 lines cut ...]
    frame #86: 0x0000000100010226 tinywhistle`runtime.run$1 at scheduler_any.go:24:10 [opt]
    frame #87: 0x000000010000fed6 tinywhistle`<goroutine wrapper> at scheduler_any.go:23:2 [opt]
    frame #88: 0x000000010000078a tinywhistle`tinygo_startTask + 6
(lldb) frame variable 
(runtime.gcBlock) root = 3678
(long) stackLen = <variable not available>

(runtime.gcBlock) block = <variable not available>

(unsigned long) start = <variable not available>

(unsigned long) end = 4311862208
(unsigned long) addr = 5385486336
(unsigned long) word = <variable not available>

(runtime.gcBlock) referencedBlock = <variable not available>

(bool) stackOverflow = <variable not available>

(runtime.gcObjectScanner) scanner = <no location, value may have been optimized out>

(lldb) q
Quitting LLDB will kill one or more processes. Do you really want to proceed: [Y/n] Y
/Volumes/git/whistle (main)$ 
```

dgryski, 2023-06-07:
I don't think that's necessarily the issue here, although highly recursive types don't play nicely with the garbage collector.  However, building with `-gc=leaking` should show it it's a GC issue or not.

deadprogram, 2023-06-07:
```
ron@ron-XPS-13-7390:~/Development/whistle (main)
$ tinygo build -o ./build/tinywhistle -gc=leaking ./cmd/whistle/
ron@ron-XPS-13-7390:~/Development/whistle (main)
$ ./build/tinywhistle ./examples/hello.lisp 
42
ron@ron-XPS-13-7390:~/Development/whistle (main)
$ ./build/tinywhistle ./examples/datalog.lisp 
panic: pattern assumed to be list
Aborted (core dumped)
```

Did I do it correctly?

deadprogram, 2023-06-07:
```
$ gdb ./build/tinywhistle ./examples/datalog.lisp 
GNU gdb (Ubuntu 9.2-0ubuntu1~20.04.1) 9.2
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./build/tinywhistle...
"/home/ron/Development/whistle/./examples/datalog.lisp" is not a core dump: file format not recognized
(gdb) c
The program is not being run.
(gdb) r
Starting program: /home/ron/Development/whistle/build/tinywhistle 
panic: pattern assumed to be list

Program received signal SIGABRT, Aborted.
__restore_sigs (set=0x7fffb800a418) at /home/ron/Development/tinygo/tinygo/lib/musl/src/signal/block.c:44
44      }
(gdb) bt
#0  __restore_sigs (set=0x7fffb800a418) at /home/ron/Development/tinygo/tinygo/lib/musl/src/signal/block.c:44
#1  0x0000000000218110 in raise (sig=<optimized out>) at /home/ron/Development/tinygo/tinygo/lib/musl/src/signal/raise.c:11
#2  0x0000000000217ede in abort () at /home/ron/Development/tinygo/tinygo/lib/musl/src/exit/abort.c:11
#3  0x000000000021d2f1 in runtime._panic (message=...) at /home/ron/Development/tinygo/tinygo/src/runtime/panic.go:52
#4  0x00000000002313ba in github.com/deosjr/whistle/lisp.unifyWithEllipsis (p=..., q=..., s=0x7fffb80ef800, depth=...) at /home/ron/Development/whistle/lisp/macro.go:240
#5  0x00000000002304ae in github.com/deosjr/whistle/lisp.syntaxRules$1 (p=...) at /home/ron/Development/whistle/lisp/macro.go:218
#6  0x0000000000233835 in github.com/deosjr/whistle/lisp.expandMacro (p=...) at /home/ron/Development/whistle/lisp/macro.go:44
#7  0x0000000000232602 in (*github.com/deosjr/whistle/lisp.process).evalEnv (p=<optimized out>, env=<optimized out>, e=...) at /home/ron/Development/whistle/lisp/env.go:70
#8  0x000000000023a3aa in (github.com/deosjr/whistle/lisp.Lisp).Eval (l=..., input=...) at /home/ron/Development/whistle/lisp/lisp.go:25
#9  0x000000000021f092 in main.startREPL () at /home/ron/Development/whistle/cmd/whistle/main.go:61
#10 main.main () at /home/ron/Development/whistle/cmd/whistle/main.go:17
#11 runtime.run$1 () at /home/ron/Development/tinygo/tinygo/src/runtime/scheduler_any.go:25
#12 <goroutine wrapper> () at /home/ron/Development/tinygo/tinygo/src/runtime/scheduler_any.go:23
#13 0x0000000000217a33 in tinygo_startTask () at /home/ron/Development/tinygo/tinygo/src/internal/task/task_stack_amd64.S:29
(gdb) l
39      }
40
41      void __restore_sigs(void *set)
42      {
43              __syscall(SYS_rt_sigprocmask, SIG_SETMASK, set, 0, _NSIG/8);
44      }
```

awmorgan, 2023-06-07:
> I don't think that's necessarily the issue here, although highly recursive types don't play nicely with the garbage collector. However, building with `-gc=leaking` should show it it's a GC issue or not.

@dgryski could you elaborate on "don't play nicely..."? 

is this `EXC_BAD_ACCESS ` on line 531 an example of this? 

the address seems so perfectly aligned that seems as if it hit an unmapped stack guard page or something: 0x141000000, or it chased a pointer to some forbidden area.

```
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x141000000)
    frame #0: 0x000000010000b67d tinywhistle`runtime.startMark(root=3678) at gc_blocks.go:531:12 [opt]
   528                  }
   529                  for addr := start; addr != end; addr += unsafe.Alignof(addr) {
   530                          // Load the word.
-> 531                          word := *(*uintptr)(unsafe.Pointer(addr))
```

deosjr, 2023-06-09:
`sdost@Sjoerds-MacBook-Pro whistle % tinygo build -opt=0 -o ./build/tinywhistle ./cmd/whistle/ 
sdost@Sjoerds-MacBook-Pro whistle % ./build/tinywhistle                                      
zsh: segmentation fault  ./build/tinywhistle
sdost@Sjoerds-MacBook-Pro whistle % tinygo build -opt=0 -gc=leaking -o ./build/tinywhistle ./cmd/whistle/
sdost@Sjoerds-MacBook-Pro whistle % ./build/tinywhistle                                                  
panic: invalid syntax (let ((pid (spawn_link (lambda () (begin (process_flag (quote eval_with_continuation) #t) (REPL env))) (quote ())))) (receive ((reason) (quasiquote (EXIT (unquote pid) (unquote reason))) -> (if (eqv? reason "normal") #t (begin (display "** exception error: ") (display reason) (display newline) (restarter env))))))
zsh: abort      ./build/tinywhistle`

Using a minimal modification to macro.go to not rely on anon funcs or closures. Panic is upon first evaluation of a macro. The problem is earlier, during parsing: I think the global macro map never gets properly populated, but it is also not empty (since that would more gracefully fail). If I understand @dgryski correctly, the above shows that there _is_ a GC issue? Which is then further obscured by compiler optimisations; the above difference doesnt show when leaving `-opt=0` out.

deadprogram, 2023-06-09:
Perhaps experiment with how you init that map https://github.com/deosjr/whistle/blob/main/lisp/macro.go#L13-L31

Perhaps try an `init()` func, and/or try to add each map entry individually to see if that has something to do with confusing the compiler.

awmorgan, 2023-06-11:
@deadprogram  I've been investigating this issue by trying to create a minimal program to replicate the error. Interestingly, adding a debugging fmt.Printf line seems to cause the tinygo compiler to crash.

```
/Volumes/git/whistle (tinygo_debug)$ tinygo build -o ./build/tinywhistle ./cmd/whistle/
panic: interp: offset out of range

goroutine 142 [running]:
github.com/tinygo-org/tinygo/interp.pointerValue.addOffset(...)
        /Users/billmorgan/git/tinygo/interp/memory.go:523
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc000a321e0, 0xc00370b540, {0xc003a71410, 0x3, 0x0?}, 0xc0010c6e10, {0xc001548ae0, 0x24})
        /Users/billmorgan/git/tinygo/interp/interpreter.go:413 +0x78c8
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc000a321e0, 0xc0030fd860, {0xc003a3c510, 0x5, 0x0?}, 0xc0010c6b10, {0xc00042d2a0, 0x20})
        /Users/billmorgan/git/tinygo/interp/interpreter.go:512 +0x7d45
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc000a321e0, 0xc00370b4f0, {0xc0039d7490, 0x7, 0x0?}, 0xc000a81c50, {0xc00042d240, 0x1c})
        /Users/billmorgan/git/tinygo/interp/interpreter.go:512 +0x7d45
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc000a321e0, 0xc00370b4a0, {0xc0039d8810, 0x8, 0x0?}, 0xc000a801b0, {0xc0039aac60, 0x18})
        /Users/billmorgan/git/tinygo/interp/interpreter.go:512 +0x7d45
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc000a321e0, 0xc00370b450, {0xc0039d6d90, 0x6, 0x0?}, 0xc0004d5080, {0xc0039aaba0, 0x14})
        /Users/billmorgan/git/tinygo/interp/interpreter.go:512 +0x7d45
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc000a321e0, 0xc0030fdf90, {0xc00394c710, 0x8, 0x0?}, 0xc0003b5b30, {0xc0039e4af0, 0x10})
        /Users/billmorgan/git/tinygo/interp/interpreter.go:512 +0x7d45
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc000a321e0, 0xc0030fdf40, {0xc00394c410, 0x8, 0x0?}, 0xc001d73c80, {0xc003999ab0, 0xc})
        /Users/billmorgan/git/tinygo/interp/interpreter.go:512 +0x7d45
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc000a321e0, 0xc0030fd1d0, {0xc00572fe10, 0x4, 0x0?}, 0xc001ec25d0, {0xc0037635f8, 0x8})
        /Users/billmorgan/git/tinygo/interp/interpreter.go:512 +0x7d45
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc000a321e0, 0xc0030fc8c0, {0x0, 0x0, 0xc003418998?}, 0x0, {0x10041ec36, 0x4})
        /Users/billmorgan/git/tinygo/interp/interpreter.go:512 +0x7d45
github.com/tinygo-org/tinygo/interp.Run({0x10024c97f?}, 0x133d05a50?, 0x0?)
        /Users/billmorgan/git/tinygo/interp/interp.go:117 +0x685
github.com/tinygo-org/tinygo/builder.optimizeProgram({0x1058051c0?}, 0xc0000c2240)
        /Users/billmorgan/git/tinygo/builder/build.go:1034 +0x3b
github.com/tinygo-org/tinygo/builder.Build.func5(0xc00312d740)
        /Users/billmorgan/git/tinygo/builder/build.go:564 +0x60d
github.com/tinygo-org/tinygo/builder.runJob(0xc00312d7a0, 0x0?)
        /Users/billmorgan/git/tinygo/builder/jobs.go:222 +0x4f
created by github.com/tinygo-org/tinygo/builder.runJobs
        /Users/billmorgan/git/tinygo/builder/jobs.go:123 +0x5be
/Volumes/git/whistle (tinygo_debug)$ 
```

deosjr, 2023-06-27:
Thank you @awmorgan , took over those changes in https://github.com/deosjr/whistle/commit/2a80ffca53ee36b9c73a5de0e26db189dbbb87e9
There is still more that doesn't work (and fails silently); see https://github.com/deosjr/whistle/issues/11#issuecomment-1584361756 for a more exhaustive list. I would think the tinygo part of this issue is (at a minimum) that the optimised build passes but fails to run? I would expect errors (which would be okay, not all go programs are expected to build under tinygo).

deadprogram, 2025-10-02:
Good news! Latest TinyGo versions are working now.

```
$ ./build/tinywhistle ./examples/datalog.lisp 
(4 3 1)
```

Closing, thanks everyone!

closed 2025-10-02 by deadprogram