tinygo-org/tinygo #3786: Import complex constants from C does not work

ysoldak, 2023-06-12:
When a C constant refers another C constant, it is not possible to use the former w/o explicitly referring the later too in Go code.

---

Reproducer, from bluetooth library
https://github.com/tinygo-org/bluetooth/blob/release/error_sd.go#L15

Replace following line
```
case e >= C.NRF_ERROR_BASE_NUM && e < C.NRF_ERROR_SDM_BASE_NUM:
```
with 
```
case e >= 0x0 && e < 0x1000:
```
And try compile, for example like this:
```
tinygo build -o test.hex -size=short -target=pca10040-s132v6 ./examples/advertisement
```

Error emitted:
```
# tinygo.org/x/bluetooth
s132_nrf52_6.1.1/s132_nrf52_6.1.1_API/include/nrf_error.h:62:48: undefined: C.NRF_ERROR_BASE_NUM
...
```

Expected outcome: constants resolved correctly w/o need for referencing them explicitly in Go code.

deadprogram, 2023-07-08:
@ysoldak is this still an issue with the latest release and/or dev branch?

aykevl, 2023-07-08:
I haven't changed anything related to CGo lately, so I'd guess yes.

deadprogram, 2025-01-13:
Since that have actually been a bunch of CGo improvements in the last couple of releases since the last time I asked, @ysoldak @aykevl is this issue still valid?

aykevl, 2025-01-15:
This should be fixed now!

ysoldak, 2025-01-16:
Let me check

ysoldak, 2025-01-16:
I confirm it works now on latest tinygo@dev(`fd89e9b83a85c0a1bfc3f6a6abebe33b7aa899e8`) and bluetooth@dev(`0b655a6f4f61075df28ea144f89721ead8be91e2`).

Now closing as fixed.

closed 2025-01-16 by ysoldak