tinygo-org/tinygo #3788: compiler crash on interface assert

awmorgan, 2023-06-12:
I originally noticed this issue while investigating another issue: https://github.com/tinygo-org/tinygo/issues/3771#issuecomment-1585869068

this is a small reproducer:

```
/Volumes/git/tinygo/x (dev)*$ uname -v
Darwin Kernel Version 20.6.0: Thu Mar  9 20:39:26 PST 2023; root:xnu-7195.141.49.700.6~1/RELEASE_X86_64
/Volumes/git/tinygo/x (dev)*$ 
/Volumes/git/tinygo/x (dev)*$ go version
go version go1.20.5 darwin/amd64
/Volumes/git/tinygo/x (dev)*$ 
/Volumes/git/tinygo/x (dev)*$ tinygo version
tinygo version 0.28.1 darwin/amd64 (using go version go1.20.5 and LLVM version 15.0.7)
/Volumes/git/tinygo/x (dev)*$ 
/Volumes/git/tinygo/x (dev)*$ cat main.go 
package main

func isStringer(i interface{}) bool {
        // Interface assert on anonymous interface type.
        _, ok := i.(interface {
                String() string
        })
        return ok
}

func main() {
        // _ = isStringer(0) // this is ok
}

func init() {
        _ = isStringer(0) // this crashes the compiler
}
/Volumes/git/tinygo/x (dev)*$ 
/Volumes/git/tinygo/x (dev)*$ tinygo build .
panic: interp: offset out of range

goroutine 39 [running]:
github.com/tinygo-org/tinygo/interp.pointerValue.addOffset(...)
        /Volumes/git/tinygo/interp/memory.go:523
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc000630640, 0xc00123ff90, {0xc000b409d0, 0x3, 0x0?}, 0xc0004acf30, {0xc00079e8a0, 0xc})
        /Volumes/git/tinygo/interp/interpreter.go:413 +0x78c8
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc000630640, 0xc00123ff40, {0xc000846490, 0x1, 0x0?}, 0xc0004ace40, {0xc00068d3b0, 0x8})
        /Volumes/git/tinygo/interp/interpreter.go:512 +0x7d45
github.com/tinygo-org/tinygo/interp.(*runner).run(0xc000630640, 0xc00123fbd0, {0x0, 0x0, 0x10001ba50?}, 0x0, {0x10041ef36, 0x4})
        /Volumes/git/tinygo/interp/interpreter.go:512 +0x7d45
github.com/tinygo-org/tinygo/interp.RunFunc({0xc0000281c0?}, 0xc0006dbce0?, 0x40?)
        /Volumes/git/tinygo/interp/interp.go:238 +0x3a5
github.com/tinygo-org/tinygo/builder.Build.func3(0xc000aa51a0)
        /Volumes/git/tinygo/builder/build.go:437 +0xcf9
github.com/tinygo-org/tinygo/builder.runJob(0xc000aa51a0, 0x0?)
        /Volumes/git/tinygo/builder/jobs.go:222 +0x4f
created by github.com/tinygo-org/tinygo/builder.runJobs
        /Volumes/git/tinygo/builder/jobs.go:123 +0x5be
/Volumes/git/tinygo/x (dev)*$ 
```

dgryski, 2023-06-14:
/cc @aykevl

awmorgan, 2023-06-18:
@dgryski  @aykevl I did some more investigation on this issue. It appears that the issue is caused by the type not having a method set.

```
package main

func main() {
}

type myint int

func init() {
	// _ = any(0).(any) // compiler crashes
	_ = any(myint(0)).(any) // compiler crashes if x() is commented below, but ok if x() is uncommented below
}

// func (i myint) x() {}
```

The panicking operation is `typecodePtr.addOffset(-int64(r.pointerSize))` in this code:

```
			case strings.HasSuffix(callFn.name, ".$typeassert"):
				if r.debug {
					fmt.Fprintln(os.Stderr, indent+"interface assert:", operands[1:])
				}

				// Load various values for the interface implements check below.
				typecodePtr, err := operands[1].asPointer(r)
				if err != nil {
					return nil, mem, r.errorAt(inst, err)
				}
				methodSetPtr, err := mem.load(typecodePtr.addOffset(-int64(r.pointerSize)), r.pointerSize).asPointer(r)
				if err != nil {
					return nil, mem, r.errorAt(inst, err)
				}
				methodSet := mem.get(methodSetPtr.index()).llvmGlobal.Initializer()
```

I think the fix is to check if the type has a method set before trying to get the pointer to the method set that doesn't exist, but I don't know how to do this.

How can I check for a valid method set in this code?

dapchen, 2023-11-02:
hit this issue as well.

deadprogram, 2024-02-28:
Completed with `v0.31.0` so now closing. Thank you!

closed 2024-02-28 by deadprogram

marco6, 2024-03-05:
Hi @deadprogram! I think it actually wasn't fixed as the original bug is still there in version 0.31.0. Given a file containing:

```go
// main.go

package main

func init() {
	_ = any(0).(interface{ x() }) // this crashes the compiler
}

func main() {}
```

Running you get:

```
$ tinygo version
tinygo version 0.31.1 linux/amd64 (using go version go1.21.1 and LLVM version 17.0.1)
$ tinygo build .
# main
<path>/main.go:6:13: interp: offset -8 out of range for object <3>
  %0 = call i1 @"interface:{main.x:func:{}{}}.$typeassert"(ptr @"reflect/types.type:basic:int"), !dbg !26

traceback:
<path>/main.go:6:13:
  %0 = call i1 @"interface:{main.x:func:{}{}}.$typeassert"(ptr @"reflect/types.type:basic:int"), !dbg !26
<path>:
  call void @"main.init#1"(ptr undef), !dbg !26
```

Right now, only the case of:

```go
_ = any(0).(any) // this crashes the compiler
```

got fixed. Can we reopen this issue? 

Last, I opened some time ago a fix to this issue. Could you take a look?

PR #4126 (2024-05-13): Checking for methodset existance
This PR adds a check for the existence of the methodset in the interpreter implementation of typeasserts. This is necessary since this [field is optional](https://github.com/tinygo-org/tinygo/blob/release/compiler/interface.go#L250).

This should be the last bit that fixes #3788 since pull #3986 only covers `XXX.(any)` cases and still crashes if the interface has methods. As such, it should also fix #4079.

As a last note, I also tried a different approach [here](https://github.com/marco6/tinygo/tree/mandatory-methodset) where I forced the methodset to be manda [... cut]