tinygo-org/tinygo #3790: Configure rp2040's usb bus (HID, VID) and detect it how sound card

csm7, 2023-06-14:
Hello guis, i use tinyGo  v0.27.0 to program WaveShare rp2040-zero. And have little problems.   Tell me out how to write code for rp2040-zero which use USB bus for transfer data between rp2040-zero and PC. The rp2040's board need to be configure (setup VID and HID of the board) to PC detect it how sound card when rp2040-zero connect to PC via USB bus. I try to use machine library  (machine.USBDev). But there are no any examples (maybe i haven't been found) which are described how to do it.

deadprogram, 2023-06-14:
Sounds like you are asking for Audio Device Class support, which is not yet available.

davecheney, 2026-10-26:
Thanks for the request. USB audio streaming is still not supported. The `machine/usb/adc/midi` package provides USB MIDI, but does not make the board appear as a sound card. Leaving this open to track USB audio support.