tinygo-org/tinygo #3792: Why machine package is called machine and not hal?

m-kru, 2023-06-14:
https://tinygo.org/docs/concepts/package-organization/

> The machine package is the TinyGo hardware abstraction layer. The interfaces that it exposes are the same regardless of the implementation of the target hardware. For example, to use whatever serial ports or timer are built-in to the target chip.

1. It is described as being actually the hal.
2. It is used in plethora of places. `hal` is much shorter than `machine`.
3. The HAL term is almost integral part of embedded systems. Anyone trying tinygo would immediately know what `hal` means, but `machine` might be surprising.

Was there any reason behind naming it `machine` instead of `hal`?

aykevl, 2023-06-16:
It's really just that MicroPython also has a `machine` library, and I was working with MicroPython at the time.
Maybe the name is a little bit confusing, but changing it now would likely be far more confusing.

m-kru, 2023-06-16:
A few finds with `-exec sed` would do 99% of the job. The major version of tinygo is still 0.

dgryski, 2023-06-16:
```
import hal "machine"
```

m-kru, 2023-06-17:
Of course you can rename imports, but this is about something else. It is about naming things what they actually are.

deadprogram, 2023-06-17:
I am not that much a fan of acronyms since they tend to augment the jargon and make things less accessible for newcomers. 

That approach and attitude is part of the success of MicroPython/CircuitPython. Not necessarily specific to the term "machine", but that term probably as well.

So as such, I do not personally feel there is an issue with TinyGo using the word "machine".

aykevl, 2023-06-17:
> A few finds with `-exec sed` would do 99% of the job. The major version of tinygo is still 0.

...except everyone using TinyGo for microcontrollers needs to update their code too. It's a big breaking change and I'm not going to do it without a very strong reason.

Yes we're at 0.x, that doesn't mean we're free to do whatever we want.

aykevl, 2023-06-17:
Closing, since I think the question has been answered sufficiently.

closed 2023-06-17 by aykevl

soypat, 2023-06-17:
I'd also comment that the machine package is not the only thing to contain the HAL. I also consider drivers to hardware to be the HAL, albeit second level HAL. 

I also agree `machine` is nice, its very clear for newcomers like Ron says.