tinygo-org/tinygo #3794: `MapIter.Key()` returns key with wrong type for maps with `any` or `interface{}` keys

fxamacker, 2023-06-16:
When compiled with TinyGo 0.28.1, `iter.Key()` returns `int` key but it should be `interface` key:

```Go
m := map[any]any{1: 2}
rv := reflect.ValueOf(m)
iter := rv.MapRange()

for iter.Next() {
  k := iter.Key()
  fmt.Printf("k.Kind() = %s\n", k.Kind())
}

// Go and TinyGo displays different results:
// - Go 1.19.10:     k.Kind() = interface
// - TinyGo 0.28.1:  k.Kind() = int
```
https://go.dev/play/p/0IkOixuogDH

This bug can cause panic in functions such as `IsNil()` by turning a value of supported type into a value of unsupported type.

dgryski, 2023-06-16:
Thanks for this bug and reproducer.  I'll look at this.

dgryski, 2023-06-16:
So I *think* this is enough to fix your issue, *but* I need to run more tests to see if this breaks anything else:
```
diff --git a/src/reflect/value.go b/src/reflect/value.go
index 96cbcb80..8b2e35a7 100644
--- a/src/reflect/value.go
+++ b/src/reflect/value.go
@@ -914,7 +914,8 @@ func (v Value) MapKeys() []Value {
        e := New(v.typecode.Elem())

        keyType := v.typecode.Key().(*rawType)
-       isKeyStoredAsInterface := keyType.Kind() != String && !keyType.isBinary()
+       keyTypeEmptyInterface := keyType.Kind() == Interface && keyType.NumMethod() == 0
+       isKeyStoredAsInterface := !keyTypeEmptyInterface && keyType.Kind() != String && !keyType.isBinary()

        for hashmapNext(v.pointer(), it, k.value, e.value) {
                if isKeyStoredAsInterface {
@@ -992,7 +993,9 @@ func (v Value) MapRange() *MapIter {
        }

        keyType := v.typecode.key()
-       isKeyStoredAsInterface := keyType.Kind() != String && !keyType.isBinary()
+
+       keyTypeEmptyInterface := keyType.Kind() == Interface && keyType.NumMethod() == 0
+       isKeyStoredAsInterface := keyType.Kind() != String && !keyType.isBinary() && !keyTypeEmptyInterface

        return &MapIter{
                m:            v,
```

PR #3795 (2023-07-02): reflect: fix iterating over maps with interface{} keys
Fixes #3794
commit ef72c5bb (2023-07-02): reflect: fix iterating over maps with interface{} keys

Fixes #3794
closed 2023-08-25 by deadprogram