tinygo-org/tinygo #3798: runtime error: nil pointer dereference (WASI)

vnaki, 2023-06-18:
wasi output error:

```
panic: runtime error: nil pointer dereference
panic: unreachable
```

TinyGo version:

```
tinygo version 0.28.1 darwin/amd64 (using go version go1.20.1 and LLVM version 15.0.0)
```

Command line:

```
tinygo build -o wasi.wasm -target=wasi main.go
```

Here 2 cases：

```golang

//export run
func run() {
  var wg = new(sync.WaitGroup)
  go func() {
     println("goroutine")
     wg.Done()
  }()
  wg.Add(1)  // here error occur
  wg.Wait()
}
```

or

``` golang
time.Sleep(time.Second)
```

Errors will occur above！

dgryski, 2023-07-04:
This is basically https://github.com/tinygo-org/tinygo/issues/2735 ; the runtime isn't initialized.

deadprogram, 2023-07-27:
@dgryski are you sure you meant #2735 ?

deadprogram, 2023-11-12:
I actually just ran into this myself, and finally understand what @dgryski was saying.

If nothing has has yet called the `_start()` function then the runtime is not yet initialized. So WASI programs that do everything from their `init()` function can run into this if they try to do anything outside the main goroutine.

Here is another example of the same problem https://github.com/fermyon/spin/issues/1259#issuecomment-1466551516

I tried the naive solution of just adding a call to `_start()` from the `init()` function here https://github.com/tinygo-org/tinygo/blob/dev/src/runtime/runtime_wasm_wasi.go#L30 but that did not appear to have any effect.

@dgryski @aykevl or anyone else, what do you think we can do about this?

aykevl, 2023-11-14:
> I tried the naive solution of just adding a call to `_start()` from the `init()` function here https://github.com/tinygo-org/tinygo/blob/dev/src/runtime/runtime_wasm_wasi.go#L30 but that did not appear to have any effect.

No that can't work. `_start` initializes the runtime, runs package initializers, and calls main. So even if `_start` was called, you would have introduced infinite recursion.

Instead, there is currently no way around the fact that you need to call `_start` first before calling any exported functions.

dgryski, 2023-11-15:
Should we come up with some supported way to initialize the runtime so that this kind of behaviour is supported?

deadprogram, 2023-11-26:
> Should we come up with some supported way to initialize the runtime so that this kind of behaviour is supported?

That would be very helpful!

deadprogram, 2025-02-21:
Is this solved now by using reactor mode?

davecheney, 2026-10-24:
Reactor mode, added in https://github.com/tinygo-org/tinygo/pull/4451, provides a supported way to initialize the runtime before calling exported functions.

This sample successfully sleeps and waits for a goroutine inside an exported call:

```go
package main

import (
	"sync"
	"time"
)

//go:wasmexport check
func check() uint32 {
	var wg sync.WaitGroup
	wg.Add(1)
	go func() {
		time.Sleep(time.Millisecond)
		wg.Done()
	}()
	wg.Wait()
	return 42
}

func main() {}
```

Save as `main.go`, then run:

```sh
tinygo build -target=wasip1 -buildmode=c-shared -o reactor.wasm main.go
wasmtime run --invoke check reactor.wasm
```

Output: `42`.

Tested with TinyGo `0.44.0-dev-eae3f875`. When embedding the module in another host, call `_initialize` once before invoking its exports. This uses reactor mode and `//go:wasmexport`, rather than calling legacy `//export` functions on an uninitialized command module.

closed 2026-10-24 by davecheney