tinygo-org/tinygo #3804: Got errors while using clang to compile the tinygo's bc file

NeoTheCapt, 2023-06-22:
```
package main

import "fmt"

func main() {
	fmt.Println("Hello World")
}
```
```
C:\tinygo\bin\tinygo.exe build -o main.bc
```
```
CC=x86_64-w64-mingw32-gcc CXX=x86_64-w64-mingw32-g++ ./clang++ /data/main.bc -o /data/main.exe -target x86_64-pc-windows-gnu -v --ld-path=/usr/x86_64-w64-mingw32/bin/ld -L /usr/lib/gcc/x86_64-w64-mingw32/10-posix -I /usr/x86_64-w64-mingw32/include
```
then I got errors:
```
root@a7a0938a650b:~/Arkari/build/bin# CC=x86_64-w64-mingw32-gcc CXX=x86_64-w64-mingw32-g++ ./clang++ /data/main.bc -o /data/main.exe -target x86_64-pc-windows-gnu -v --ld-path=/usr/x86_64-w64-mingw32/bin/ld -L /usr/lib/gcc/x86_64-w64-mingw32/10-posix -I /usr/x86_64-w64-mingw32/include
clang version 16.0.5 (https://github.com/KomiMoe/Arkari.git 82aaa4b2327e042bac257eeb26859bc485bea0df)
Target: x86_64-pc-windows-gnu
Thread model: posix
InstalledDir: /root/Arkari/build/bin/.
clang-16: warning: argument unused during compilation: '-I /usr/x86_64-w64-mingw32/include' [-Wunused-command-line-argument]
 "/root/Arkari/build/bin/clang-16" -cc1 -triple x86_64-pc-windows-gnu -emit-obj -mrelax-all -disable-free -clear-ast-before-backend -main-file-name main.bc -mrelocation-model pic -pic-level 2 -mframe-pointer=none -fmath-errno -ffp-contract=on -fno-rounding-math -mconstructor-aliases -mms-bitfields -funwind-tables=2 -target-cpu x86-64 -tune-cpu generic -mllvm -treat-scalable-fixed-error-as-warning -debugger-tuning=gdb -v -fcoverage-compilation-dir=/root/Arkari/build/bin -resource-dir /root/Arkari/build/lib/clang/16 -fdebug-compilation-dir=/root/Arkari/build/bin -ferror-limit 19 -fno-use-cxa-atexit -fgnuc-version=4.2.1 -exception-model=seh -fcolor-diagnostics -faddrsig -o /tmp/main-d60c29.o -x ir /data/main.bc
clang -cc1 version 16.0.5 based upon LLVM 16.0.5 default target x86_64-unknown-linux-gnu
warning: overriding the module target triple with x86_64-pc-windows-gnu [-Woverride-module]
/data/main.bc

1 warning generated.
 "/usr/x86_64-w64-mingw32/bin/ld" -m i386pep -Bdynamic -o /data/main.exe /usr/x86_64-w64-mingw32/lib/crt2.o /usr/x86_64-w64-mingw32/lib/crtbegin.o -L/usr/lib/gcc/x86_64-w64-mingw32/10-posix -L/usr/x86_64-w64-mingw32/lib -L/usr/x86_64-w64-mingw32/mingw/lib /tmp/main-d60c29.o -lstdc++ -lmingw32 -lgcc_s -lgcc -lmoldname -lmingwex -lmsvcrt -ladvapi32 -lshell32 -luser32 -lkernel32 -lmingw32 -lgcc_s -lgcc -lmoldname -lmingwex -lmsvcrt -lkernel32 /usr/x86_64-w64-mingw32/lib/crtend.o
/usr/x86_64-w64-mingw32/bin/ld: /tmp/main-d60c29.o: in function `mainCRTStartup':
C:\tinygo\src\runtime/runtime_windows.go:(.text+0x82d0): multiple definition of `mainCRTStartup'; /usr/x86_64-w64-mingw32/lib/crt2.o:./build/x86_64-w64-mingw32-x86_64-w64-mingw32-crt/./mingw-w64-crt/crt/crtexe.c:206: first defined here
/usr/x86_64-w64-mingw32/bin/ld: /tmp/main-d60c29.o: in function `(*internal/task.state).pause':
C:\tinygo\src\internal\task/task_stack_amd64_windows.go:78: undefined reference to `tinygo_swapTask'
/usr/x86_64-w64-mingw32/bin/ld: /tmp/main-d60c29.o: in function `runtime.markStack':
C:\tinygo\src\runtime/gc_stack_raw.go:14: undefined reference to `tinygo_scanCurrentStack'
/usr/x86_64-w64-mingw32/bin/ld: /tmp/main-d60c29.o: in function `runtime._panic':
C:\tinygo\src\runtime/panic.go:45: undefined reference to `tinygo_longjmp'
/usr/x86_64-w64-mingw32/bin/ld: /tmp/main-d60c29.o: in function `(*internal/task.state).resume':
C:\tinygo\src\internal\task/task_stack_amd64_windows.go:72: undefined reference to `tinygo_swapTask'
/usr/x86_64-w64-mingw32/bin/ld: /tmp/main-d60c29.o: in function `os.runtime_args':
C:\tinygo\src\runtime/runtime_windows.go:72: undefined reference to `_configure_narrow_argv'
/usr/x86_64-w64-mingw32/bin/ld: C:\tinygo\src\runtime/runtime_windows.go:73: undefined reference to `__p___argc'
/usr/x86_64-w64-mingw32/bin/ld: /tmp/main-d60c29.o:main:(.rdata$.refptr.tinygo_startTask[.refptr.tinygo_startTask]+0x0): undefined reference to `tinygo_startTask'
/usr/x86_64-w64-mingw32/bin/ld: /usr/x86_64-w64-mingw32/lib/libmingw32.a(lib64_libmingw32_a-crt0_c.o): in function `main':
./build/x86_64-w64-mingw32-x86_64-w64-mingw32-crt/./mingw-w64-crt/crt/crt0_c.c:18: undefined reference to `WinMain'
clang-16: error: linker command failed with exit code 1 (use -v to see invocation)
```

aykevl, 2023-07-05:
Why are you doing this?

Compiling to bitcode mostly exists for debugging purposes and is not generally supported. Note that TinyGo doesn't need an installed cross compiler, you can simply use GOOS=windows from any operating system.

aykevl, 2023-09-05:
I don't think there is anything to fix here, so closing.

closed 2023-09-05 by aykevl