tinygo-org/tinygo #3822: Help with Flashing Error

AcousticTypewriter, 2023-07-05:
Hey there!

I'm a new tinygo user, and I'm trying to flash my first program onto my RPI Pico. I installed go version 1.19.8 and tiny go version 0.28.1. I started off copying a blinky led program and tried to flash it. With this command,

tinygo flash -target-pico ./main.go

I get this error:

error: failed to flash /tmp/tinygo1032452387/main.uf2: unable to locate any volume: [RPI-RP2]

I've tried searching on google and through this repository and have not found any information relating to this error. I am running debian on a chromebook. Hopefully someone else has already got tinygo running on the chromebook linux distro. 

Thanks in advance for any advice!

AcousticTypewriter, 2023-07-05:
I've delved deeper into the issue and I've discovered that chromeos just puts the pico into a weird place in the fs. It puts the pico into /mnt/chromeos/removable/, but I guess tinygo doesn't search there. Is there any way to point the tinygo flash command to this folder to find the pico?

Thanks!

AcousticTypewriter, 2023-07-06:
So I've found the piece of source code that finds the volume to flash the uf2 to, and it definitely doesn't look at /mnt/chromeos/removable/ which means that it will not discover anything on chromeos. Is this something that I would have to make a fork/pull request to get chromeos support or is there some other way to change where it looks?

Referenced source code: https://github.com/tinygo-org/tinygo/blob/d01d85930d19a73010853b7ba8648d8811891bee/main.go#L948

AcousticTypewriter, 2023-07-06:
While I still think this support would be very good and I would be fully able to implement this, I was able to solve this issue with a few aliases and symbolic links. Here's the aliases:

alias tgoinit='sudo unlink /media/removable && sudo ln -s /mnt/chromeos/removable/ /media'
alias tgo='tgoinit && tinygo flash -target=pico ./main.go'

Make sure that you share the RPI folder with linux, and then don't connect the RP2 Boot to linux. Wait 5ish seconds, and then you can run tgoinit and you should be up and running.

Love to answer any questions as I spent excessive amounts of time on this topic.
Peace!

closed 2023-07-06 by AcousticTypewriter

deadprogram, 2023-07-07:
@AcousticTypewriter this might be addressed by https://github.com/tinygo-org/tinygo/pull/3816 which is now in the `dev` branch.

If you got a chance to take a look to see if it does, it would be very appreciated.