tinygo-org/tinygo #3834: WASI problem

Rui-Gan, 2023-07-14:
I am trying to compile the eKuiper project(https://github.com/lf-edge/ekuiper) to Wasi and then use wasmedge to run the compiled binary file. I used TinyGo to compile eKuiper to Wasi, but encountered issues.

When I ran the command below to compile eKuiper to Wasi:
```
tinygo build -o kuperd.wasm -target wasi cmd/kuiperd/main.go
```
I received the following error message:
```
# golang.org/x/sys/unix
../../../go/pkg/mod/golang.org/x/sys@v0.10.0/unix/syscall_linux.go:1901:64: undefined: syscall.Rlimit
../../../go/pkg/mod/golang.org/x/sys@v0.10.0/unix/syscall_linux.go:1906:50: undefined: syscall.Rlimit
../../../go/pkg/mod/golang.org/x/sys@v0.10.0/unix/syscall_linux.go:1906:79: undefined: syscall.Rlimit
../../../go/pkg/mod/golang.org/x/sys@v0.10.0/unix/syscall_unix.go:595:17: undefined: syscall.Setrlimit
../../../go/pkg/mod/golang.org/x/sys@v0.10.0/unix/syscall_unix.go:595:47: undefined: syscall.Rlimit
```

I am not sure what causes this error and would appreciate any help. I would like to know why this issue occurs and how to resolve it.

HarikrishnanBalagopal, 2023-07-19:
> syscall.Rlimit

Recursion limit/max call stack size I think https://pkg.go.dev/syscall#Rlimit

Rui-Gan, 2023-07-20:
Thanks for your reply,

closed 2023-07-20 by Rui-Gan

aykevl, 2023-08-11:
Also see: #3861

aykevl, 2023-08-12:
You can try `tinygo list -deps -json -target wasi` and inspect the output.

inliquid, 2023-09-28:
Also encountered this issue with some gRPC (proto definitions) dependencies.

aykevl, 2023-09-29:
You can try `GOOS=wasip1 GOARCH=wasm tinygo ...` instead of `tinygo -target=wasi`. This means we use the standard library WASI support instead of our custom variant, which _might_ get more code to compile.

inliquid, 2023-09-29:
Thank you, will this approach become a standard way of compiling to WASI at some point?

aykevl, 2023-10-01:
Yes. It only works in Go 1.21+ and right now we support Go versions down to Go 1.18 so I think we should wait a bit before changing the default.