tinygo-org/tinygo #3835: Exclude `.vscode` folder from source control

soypat, 2023-07-15:
I understand many developers here use VSCode as their daily driver IDE.

Could we add a line to `.gitignore` to exclude the settings directory from being tracked when committing changes?

Also, props to whatever change allowed us to work inside the tinygo repo with intellisense! It's been such a boon to my productivity!

soypat, 2023-07-18:
I had no idea configuring gitignores globally was possible. For future reference this is done by adding the file to git's global config:
```
git config --global core.excludesfile $HOME/.gitignore
```

Marking issue as resolved/closed.

soypat, 2023-07-18:
> Apparently it does now? I didn't really check.

@aykevl  Does on my end!
[image]

aykevl, 2023-07-18:
> For future reference this is done by adding the file to git's global config:

Oh, I see, I forgot I added that file in my `.gitconfig` file.

For locally ignored files (in a specific checkout), you can also modify `.git/info/exclude`. That's like `.gitignore` but won't be committed.