tinygo-org/tinygo #3839: `//go:wasm-module` doesn't seem to work

Mossaka, 2023-07-21:
> ... The Wasm module (which has a default value of `env`) can be specified using `//go:wasm-module <module>`.

The `//go:wasm-module <module-name>` directive supposed to change the module name to `<module-name>` given by the user. However, the produced Wasm module has a different naming scheme for the module name. 

## Code

```Go
package main

//go:wasm-module mymodule
//go:export add
func add(x, y uint32) uint32 {
	return x + y
}

// main is required for the `wasi` target, even if it isn't used.
func main() {}

```

## Wasm target

1. Build the wasm module by running `tinygo build -target=wasm -o go_example.wasm main.go`
2. Check the exported function signature by running `wasm-tools print go_example.wasm | grep export`

You will see something like

```Wasm
(export ...)
(export "main.add" (func $main.add.command_export))
(export ...)
```

## WASI target

1. Build the wasm module by running `tinygo build -target=wasm -o go_example.wasm main.go`
2. Check the exported function signature by running `wasm-tools print go_example.wasm | grep export`

You will see something like

```Wasm
(export "main.add" (func $main.add))
```

## Expected Result

```Wasm
(export "mymodule" (func $main.add))
```

jordan-rash, 2023-07-27:
I just ran into this issue as well.  It seems no matter what module name it is given, its compiled to "package.name"

aykevl, 2023-07-28:
Looking into it. This issue is present at least from TinyGo 0.27, probably earlier too.

aykevl, 2023-07-28:
Here is a fix: https://github.com/tinygo-org/tinygo/issues/3839

jordan-rash, 2023-07-28:
I pull this and it is still not working for me.

jordan-rash, 2023-07-28:
More context as to where this request is originating, using BCA tooling and given the following `.wit` file, you can see the export that `component new` is looking for -> `example:host/derp#hostrun`

Using my previous patch, i was able to achieve that by 
```
//go:wasm-module example:host/derp
//export hostrun
```

I am not advocating for my hack, just simply trying to share what I was doing when the problem came up.

```
└─❯ cat wit/host.wit                                                                                                                    
package example:host

interface derp {
  hostrun: func()
}

world host {
  import print: func(msg: string)

  export derp
}

└─❯ wasm-tools component embed --world host ./wit main.wasm -o main.embed.wasm                                                          

└─❯ wasm-tools component new main.embed.wasm --adapt wasi_snapshot_preview1=./wasi_snapshot_preview1.command.wasm -o main.component.wasm
error: failed to encode a component from module

Caused by:
    0: failed to validate exported interface `example:host/derp`
    1: module does not export required function `example:host/derp#hostrun`
```

aykevl, 2023-07-28:
```go
//go:wasm-module mymodule
//export guestcall
func guestcall(operationSize uint32, payloadSize uint32) bool { // nolint
```
What do you expect to happen here? It doesn't make sense to put a `//go:wasm-module` on an _exported_ function, only on imported functions. You are building a module, with a particular name, and each function only has a function name (not a module name).

jordan-rash, 2023-07-28:
> If you really want to name the function example:host/derp#hostrun, what about //export example:host/derp#hostrun?

I have to apologize, I did try this before I posted and it didn't work.  I must have been on the wrong branch because now I am getting the naming I expected.  Thank you very much. 

And yes, I agree with all your comments, but I was misunderstanding some of the build tags.  I still new to this 😅 

That said, I can confirm the fix worked for me.  Again, thank you

aykevl, 2023-07-31:
> I have to apologize, I did try this before I posted and it didn't work.

Before my fix the `//go:wasm-module` would have broken the export, so that makes sense.

I'm glad it works for you now!

deadprogram, 2023-08-26:
Closing since this is part of release 0.29.0 thank you everyone!

closed 2023-08-26 by deadprogram