tinygo-org/tinygo #3842: panic: unknown struct {}{}:{}

stalecu, 2023-07-25:
Hello! I am new to TinyGo, so I decided to try it on my trusty RP2040. I cloned the repo on the release branch (which I suppose means I am on 0.28.1) to a separate repository, built LLVM myself (it didn't want to comply with my system LLVM-15 on openSUSE Tumbleweed, but oh well) and then symlinked tinygo to /usr/bin. Then I created a new project (module?) and this is the contents of hello.mod:

```go
package main

import (
    "machine"
    "time"
)

func main() {
    led := machine.LED
    config := machine.PinConfig{
        Mode: machine.PinOutput,
    }

    led.Configure(config)
    for {
        led.Low()
        time.Sleep(time.Millisecond * 500)

        led.High()
        time.Sleep(time.Millisecond * 500)
    }
}
``` 

and this is what I get when building with `tinygo build -o blink -target="pico" ./hello.go` (to simplify the output a bit, I replaced my actual tinygo path with `$TINYGO_PATH`:
```
panic: unknown constant: struct{}{}:struct{}

goroutine 55 [running]:
github.com/tinygo-org/tinygo/compiler.(*compilerContext).createConst(0xc008a34000, 0xc008a60340, 0xc0059c7310?)
	$TINYGO_PATH/compiler/compiler.go:2972 +0xd74
github.com/tinygo-org/tinygo/compiler.(*builder).getValue(0xc00acdb680, {0x59c41b0?, 0xc008a60340?}, 0xc0059c74e0?)
	$TINYGO_PATH/compiler/compiler.go:1901 +0xcf
github.com/tinygo-org/tinygo/compiler.(*builder).createInstruction(0xc00acdb680, {0x59c4af8?, 0xc008a429c0?})
	$TINYGO_PATH/compiler/compiler.go:1494 +0x368
github.com/tinygo-org/tinygo/compiler.(*builder).createFunction(0xc00acdb680)
	$TINYGO_PATH/compiler/compiler.go:1349 +0xaa5
github.com/tinygo-org/tinygo/compiler.(*compilerContext).createPackage(0xc008a34000, {0xc008974480?}, 0xc003c13e00)
	$TINYGO_PATH/compiler/compiler.go:865 +0x345
github.com/tinygo-org/tinygo/compiler.CompilePackage({0xc00002b467?, 0x5a?}, 0xc00042e960, 0xc003c13e00, {0xcea4cdf91292f0e3?}, 0x593cb42?, 0xea?)
	$TINYGO_PATH/compiler/compiler.go:312 +0x459
github.com/tinygo-org/tinygo/builder.Build.func3(0xc00601e780)
	$TINYGO_PATH/builder/build.go:357 +0x20e
github.com/tinygo-org/tinygo/builder.runJob(0xc00601e780, 0x0?)
	/home/a13cui/code/tinygo/builder/jobs.go:222 +0x4f
created by github.com/tinygo-org/tinygo/builder.runJobs
	/home/a13cui/code/tinygo/builder/jobs.go:123 +0x5be
```

Is there something I'm missing? And perhaps a way to reinstall everything from the tarball if that might be the culprit. Thank you for your attention.

stalecu, 2023-07-25:
Edit: I nuked everything and tried installing it from the tarball... maybe LLVM was acting weird? I don't know.

closed 2023-07-25 by stalecu

QuLogic, 2023-08-15:
> `panic: unknown constant: struct{}{}:struct{}`

It seems like you may have compiled against golang.org/x/tools >= 0.9.0. This version started lifting struct and array constants since https://github.com/golang/tools/commit/71ea8f168c160da91116b4ddbd577a8fc95aa334 and the compiler doesn't handle that in `createConst` yet.

QuLogic, 2023-09-23:
This would be fixed by #3883.