tinygo-org/tinygo #3862: Unimplemented reflect.Value MethodByName()

kefniark, 2023-08-11:
## Description
I was tried to build one of the ebitengine example with tinygo (nothing complicated, the sample from [their getting started](https://ebiten-zh.vercel.app/documents/install.html)).
```sh
go get github.com/hajimehoshi/ebiten/v2
go run -tags=example github.com/hajimehoshi/ebiten/v2/examples/rotate
```

The default generated wasm binary size is quite big, and tinygo seem to do a better job for that.
So I gave a try and the build works fine, and the size seems quite promising.
Sadly I'm running into some reflection issue with the generated wasm:
```
panic: unimplemented: (reflect.Value).MethodByName()
yourgame.tiny.wasm:0x211b Uncaught (in promise) RuntimeError: unreachable
    at runtime._panic (yourgame.tiny.wasm:0x211b)
    at (*text/template.state).evalField (yourgame.tiny.wasm:0xf11b2)
    at (*text/template.state).evalFieldChain (yourgame.tiny.wasm:0xeec1a)
    at (*text/template.state).evalFieldNode (yourgame.tiny.wasm:0xeea29)
    at (*text/template.state).evalPipeline (yourgame.tiny.wasm:0xef459)
    at (*text/template.state).walkIfOrWith (yourgame.tiny.wasm:0xf80e3)
    at (*text/template.state).walk (yourgame.tiny.wasm:0xf7106)
    at (*text/template.state).walk (yourgame.tiny.wasm:0xf71a1)
    at github.com/hajimehoshi/ebiten/v2/internal/builtinshader.Shader (yourgame.tiny.wasm:0xf8b00)
    at byn$mgfn-shared$github.com/hajimehoshi/ebiten/v2/internal/restorable.init#1$1 (yourgame.tiny.wasm:0x143a0e)
```

## Versions
* Ebitengine 2.5.6
* Go 1.20
* Tinygo 1.28.1

aykevl, 2023-08-11:
Unfortunately we don't support method names in reflect at the moment.

butaca, 2024-04-16:
It would be amazing if this worked because the wasm size is incredible smaller with tinygo which is vital for browser games. The wasm size generated by regular go is too big, even when brotlied.

Distortions81, 2025-02-09:
Would love to see this.

ymotongpoo, 2026-10-07:
`reflect.Type.MethodByName` has the same gap, and it blocks protobuf-go. `proto.Marshal` panics on the first call for any generated message.

```go
package main

import (
	"time"

	"google.golang.org/protobuf/proto"
	"google.golang.org/protobuf/types/known/wrapperspb"
)

func main() {
	time.Sleep(2 * time.Second)
	b, err := proto.Marshal(wrapperspb.String("hello"))
	println(len(b), err == nil)
}
```

With google.golang.org/protobuf v1.36.12, `go run .` prints `7 true`. TinyGo 0.42.0 builds and links the program, but it panics at run time. I see the same output with `tinygo run .` on linux/amd64 and on an ESP32-S3 (`esp32s3-generic`).

```
panic: unimplemented: (reflect.Type).MethodByName()
```

The call comes from `makeStructInfo` in `internal/impl/message.go`, lines 198 to 205. It looks up `XXX_OneofFuncs` and `XXX_OneofWrappers` on every message type, even when the message has no oneof. Both names are constants. When a lookup succeeds, protobuf-go calls the method through `Method.Func.Call`. So protobuf-go needs `Type.MethodByName`, and it also needs `Call` on the returned `Func`.