tinygo-org/tinygo #3871: "TypeError: wasm function signature contains illegal type" error on current dev branch

deadprogram, 2023-08-16:
Running `make test` using Go 1.21 using the latest `dev` branch results in errors:

```
tinygo:ld.lld: error: undefined symbol: notARealFunction                                                                                                                     
>>> referenced by builderr.go:9 (/home/ron/Development/tinygo/tinygo/tests/testing/builderr/builderr.go:9)                                                                   
>>>               /home/ron/.cache/tinygo/thinlto/llvmcache-4457084472EE654FE0B37BA07A9F7F0BFC5D7957:(github.com/tinygo-org/tinygo/tests/testing/builderr_test.TestThing)    
failed to run tool: ld.lld                                                                                                                                                   
TypeError: wasm function signature contains illegal type                                                                                                                     
    at (syscall/js.Value).Get (<anonymous>:wasm-function[142]:0x12a0a)                                                                                                       
    at runtime.run$1 (<anonymous>:wasm-function[180]:0x18dc8)                                                                                                                
    at runtime.run$1$gowrapper (<anonymous>:wasm-function[130]:0x111bc)                                                                                                      
    at tinygo_launch (<anonymous>:wasm-function[14]:0x5f7)                                                                                                                   
    at (*internal/task.Task).Resume (<anonymous>:wasm-function[77]:0x8ed2)                                                                                                   
    at runtime.scheduler (<anonymous>:wasm-function[131]:0x11274)                                                                                                            
    at _start (<anonymous>:wasm-function[128]:0x110e7)
    at _start.command_export (<anonymous>:wasm-function[415]:0x579f4)
    at global.Go.run (/home/ron/Development/tinygo/tinygo/targets/wasm_exec.js:480:24) 
    at /home/ron/Development/tinygo/tinygo/targets/wasm_exec.js:523:14
TypeError: wasm function signature contains illegal type
    at (syscall/js.Value).Get (<anonymous>:wasm-function[142]:0x12a0c)
    at runtime.run$1 (<anonymous>:wasm-function[180]:0x18dca)
    at runtime.run$1$gowrapper (<anonymous>:wasm-function[130]:0x111be)
    at tinygo_launch (<anonymous>:wasm-function[14]:0x5f9)
    at (*internal/task.Task).Resume (<anonymous>:wasm-function[77]:0x8ed4)
    at runtime.scheduler (<anonymous>:wasm-function[131]:0x11276)
    at _start (<anonymous>:wasm-function[128]:0x110e9)
    at _start.command_export (<anonymous>:wasm-function[417]:0x5833f)
    at global.Go.run (/home/ron/Development/tinygo/tinygo/targets/wasm_exec.js:480:24) 
    at /home/ron/Development/tinygo/tinygo/targets/wasm_exec.js:523:14
--- FAIL: TestTest (0.00s)
    --- FAIL: TestTest/WASM (0.00s)
        --- FAIL: TestTest/WASM/Pass (19.98s)
            main_test.go:531: FAIL      github.com/tinygo-org/tinygo/tests/testing/pass 1.107s
            main_test.go:443: test failed
```

kenbell, 2023-08-16:
FWIW, I can't repro:
```sh
$ go clean -testcache
$ tinygo version
tinygo version 0.29.0-dev-f4375d04 linux/amd64 (using go version go1.21.0 and LLVM version 15.0.0)
$ make test
CGO_CPPFLAGS="-I/home/kenneth/work/tinygo/main/llvm-project/llvm/include -I/home/kenneth/work/tinygo/main/llvm-build/include -D_GNU_SOURCE -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -D__STDC_LIMIT_MACROS -I/home/kenneth/work/tinygo/main/llvm-build/tools/clang/include -I/home/kenneth/work/tinygo/main/llvm-project/clang/include -I/home/kenneth/work/tinygo/main/llvm-project/lld/include" CGO_CXXFLAGS="-std=c++14" CGO_LDFLAGS="-L/home/kenneth/work/tinygo/main/llvm-build/lib -lclang -Wl,--start-group -lclangAnalysis -lclangAST -lclangASTMatchers -lclangBasic -lclangCodeGen -lclangCrossTU -lclangDriver -lclangDynamicASTMatchers -lclangEdit -lclangExtractAPI -lclangFormat -lclangFrontend -lclangFrontendTool -lclangHandleCXX -lclangHandleLLVM -lclangIndex -lclangLex -lclangParse -lclangRewrite -lclangRewriteFrontend -lclangSema -lclangSerialization -lclangSupport -lclangTooling -lclangToolingASTDiff -lclangToolingCore -lclangToolingInclusions -Wl,--end-group -lstdc++ -Wl,--start-group -llldCOFF -llldCommon -llldELF -llldMachO -llldMinGW -llldWasm -Wl,--end-group -L/home/kenneth/work/tinygo/main/llvm-build/lib  -lLLVMWindowsManifest -lLLVMWindowsDriver -lLLVMMCJIT -lLLVMLTO -lLLVMPasses -lLLVMObjCARCOpts -lLLVMExtensions -lLLVMLibDriver -lLLVMOption -lLLVMInterpreter -lLLVMExecutionEngine -lLLVMRuntimeDyld -lLLVMOrcTargetProcess -lLLVMOrcShared -lLLVMCoverage -lLLVMCoroutines -lLLVMXtensaDisassembler -lLLVMXtensaAsmParser -lLLVMXtensaCodeGen -lLLVMXtensaDesc -lLLVMXtensaInfo -lLLVMAVRDisassembler -lLLVMAVRAsmParser -lLLVMAVRCodeGen -lLLVMAVRDesc -lLLVMAVRInfo -lLLVMWebAssemblyDisassembler -lLLVMWebAssemblyAsmParser -lLLVMWebAssemblyCodeGen -lLLVMWebAssemblyDesc -lLLVMWebAssemblyUtils -lLLVMWebAssemblyInfo -lLLVMRISCVDisassembler -lLLVMRISCVAsmParser -lLLVMRISCVCodeGen -lLLVMRISCVDesc -lLLVMRISCVInfo -lLLVMAArch64Disassembler -lLLVMAArch64AsmParser -lLLVMAArch64CodeGen -lLLVMAArch64Desc -lLLVMAArch64Utils -lLLVMAArch64Info -lLLVMARMDisassembler -lLLVMARMAsmParser -lLLVMARMCodeGen -lLLVMipo -lLLVMVectorize -lLLVMLinker -lLLVMIRReader -lLLVMAsmParser -lLLVMFrontendOpenMP -lLLVMARMDesc -lLLVMARMUtils -lLLVMARMInfo -lLLVMX86TargetMCA -lLLVMMCA -lLLVMX86Disassembler -lLLVMX86AsmParser -lLLVMX86CodeGen -lLLVMCFGuard -lLLVMGlobalISel -lLLVMX86Desc -lLLVMX86Info -lLLVMMCDisassembler -lLLVMSelectionDAG -lLLVMInstrumentation -lLLVMAsmPrinter -lLLVMCodeGen -lLLVMTarget -lLLVMScalarOpts -lLLVMInstCombine -lLLVMAggressiveInstCombine -lLLVMTransformUtils -lLLVMBitWriter -lLLVMAnalysis -lLLVMProfileData -lLLVMSymbolize -lLLVMDebugInfoPDB -lLLVMDebugInfoMSF -lLLVMDebugInfoDWARF -lLLVMObject -lLLVMTextAPI -lLLVMMCParser -lLLVMMC -lLLVMDebugInfoCodeView -lLLVMBitReader -lLLVMCore -lLLVMRemarks -lLLVMBitstreamReader -lLLVMBinaryFormat -lLLVMSupport -lLLVMDemangle -lrt -ldl -lm -lstdc++ " go test  -timeout=20m -buildmode exe -tags "byollvm osusergo" ./builder ./cgo ./compileopts ./compiler ./interp ./transform .
ok      github.com/tinygo-org/tinygo/builder    2.300s
ok      github.com/tinygo-org/tinygo/cgo        0.026s
ok      github.com/tinygo-org/tinygo/compileopts        0.003s
ok      github.com/tinygo-org/tinygo/compiler   1.499s
ok      github.com/tinygo-org/tinygo/interp     0.006s
ok      github.com/tinygo-org/tinygo/transform  0.084s
ok      github.com/tinygo-org/tinygo    19.071s
```

kenbell, 2023-08-16:
The TypeError stuff and lld.ld is I think noise - just a test that's meant to fail, failing.  It looks like `TestTest/WASM/Pass` is the real culprit.

Just in case it's threading/timing related, it looks like `TestTest` in `main_test.go` has a bunch of `t.Parallel()` calls - worth commenting them all out?

deadprogram, 2023-08-17:
@kenbell for sure your machine runs those tests a LOT faster than mine:

```
ok      github.com/tinygo-org/tinygo/builder    6.101s                                                                                                                       
ok      github.com/tinygo-org/tinygo/cgo        0.061s                                                                                                                       
ok      github.com/tinygo-org/tinygo/compileopts        0.011s                                                                                                               
ok      github.com/tinygo-org/tinygo/compiler   5.600s                                                                                                                       
ok      github.com/tinygo-org/tinygo/interp     0.012s                                                                                                                       
ok      github.com/tinygo-org/tinygo/transform  0.164s
```

deadprogram, 2023-08-17:
> Just in case it's threading/timing related, it looks like TestTest in main_test.go has a bunch of t.Parallel() calls - worth commenting them all out?

I tried that but same problem.

dkegel-fastly, 2023-08-17:
Same problem with go 1.20.7 here.

dkegel-fastly, 2023-08-18:
git bisect says (drum roll please):
```
commit 4d2a6d2bbeef8db2a2e9714f3d91372f89bc9b80 (HEAD)
Author: Ayke van Laethem <aykevanlaethem@gmail.com>
Date:   Sun Apr 30 17:54:03 2023 +0200

    wasm: remove i64 workaround, use BigInt instead
```

dkegel-fastly, 2023-08-18:
So, it's an old node version.

The developer doc and perhaps the makefile should probably make the node >= 16.x version requirement more obvious...?

deadprogram, 2023-08-18:
I just figured out the exact same thing, then saw your comment. We're so synced up :infinity:

deadprogram, 2023-08-18:
> The developer doc and perhaps the makefile should probably make the node >= 16.x version requirement more obvious...?

Yes, agreed. Looking at `Makefile` now...

deadprogram, 2023-08-26:
Closing since this is part of release 0.29.0 thank you everyone!

PR #3875 (2023-08-19): make: add task to check NodeJS version before running tests
This PR adds a check to the `Makefile` to verify that NodeJS v16+ is installed before running tests, since it is required by those tests.

Fixes #3871
closed 2023-08-26 by deadprogram