tinygo-org/tinygo #3874: Compiler crash

rockwotj, 2023-08-18:
We had a compiler crash during a CI run, however it's the only time this has happened, so maybe the stacktrace is helpful, but note this is likely some race condition somewhere, as this generally doesn't happen.

```
/vectorized/bin/tinygo build -o /var/lib/buildkite-agent/builds/buildkite-amd64-builders-i-045ec208102639ae1-1/redpanda/redpanda/vbuild/release/clang/src/go/transform-sdk/internal/testdata/wasi.wasm -target wasi wasi/transform.go
fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x80 addr=0x0 pc=0x5e62a2d]
runtime stack:
runtime.throw({0x8ebde5?, 0x7fb81d489fc0?})
/usr/local/go/src/runtime/panic.go:1047 +0x5d fp=0x7fb822e822f0 sp=0x7fb822e822c0 pc=0x255b0fd
runtime.sigpanic()
/usr/local/go/src/runtime/signal_unix.go:825 +0x3e9 fp=0x7fb822e82350 sp=0x7fb822e822f0 pc=0x25723a9
goroutine 152 [syscall]:
runtime.cgocall(0x28e2670, 0xc005183b30)
/usr/local/go/src/runtime/cgocall.go:157 +0x5c fp=0xc005183b08 sp=0xc005183ad0 pc=0x252a89c
tinygo.org/x/go-llvm._Cfunc_LLVMVerifyModule(0x7fb8258b0330, 0x1, 0xc006104130)
_cgo_gotypes.go:9315 +0x4c fp=0xc005183b30 sp=0xc005183b08 pc=0x276292c
tinygo.org/x/go-llvm.VerifyModule.func1({0x276a93f?}, 0x258b0330?, 0x7fb82503b610?)
/home/rockwood/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20221028183034-8341240c0b32/analysis.go:38 +0x7d fp=0xc005183b78 sp=0xc005183b30 pc=0x2762e5d
tinygo.org/x/go-llvm.VerifyModule({0xc00043c446?}, 0x53?)
/home/rockwood/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20221028183034-8341240c0b32/analysis.go:38 +0x45 fp=0xc005183bd0 sp=0xc005183b78 pc=0x2762d25
github.com/tinygo-org/tinygo/builder.Build.func3(0xc003fa6480)
/home/rockwood/Workspace/tinygo/builder/build.go:363 +0x397 fp=0xc005183f78 sp=0xc005183bd0 pc=0x28677b7
github.com/tinygo-org/tinygo/builder.runJob(0xc003fa6480, 0x0?)
/home/rockwood/Workspace/tinygo/builder/jobs.go:222 +0x4f fp=0xc005183fc0 sp=0xc005183f78 pc=0x2873a4f
github.com/tinygo-org/tinygo/builder.runJobs.func2()
/home/rockwood/Workspace/tinygo/builder/jobs.go:123 +0x2a fp=0xc005183fe0 sp=0xc005183fc0 pc=0x287338a
runtime.goexit()
/usr/local/go/src/runtime/asm_amd64.s:1598 +0x1 fp=0xc005183fe8 sp=0xc005183fe0 pc=0x258d8a1
created by github.com/tinygo-org/tinygo/builder.runJobs
/home/rockwood/Workspace/tinygo/builder/jobs.go:123 +0x5be
goroutine 1 [chan receive]:
runtime.gopark(0xc0005caae0?, 0xc0047441e0?, 0xa0?, 0xd9?, 0xc0050f65f0?)
/usr/local/go/src/runtime/proc.go:381 +0xd6 fp=0xc0050f6580 sp=0xc0050f6560 pc=0x255de56
runtime.chanrecv(0xc0014781e0, 0xc0050f6738, 0x1)
/usr/local/go/src/runtime/chan.go:583 +0x49d fp=0xc0050f6610 sp=0xc0050f6580 pc=0x252d65d
runtime.chanrecv1(0x8713c0?, 0xc003faea20?)
/usr/local/go/src/runtime/chan.go:442 +0x18 fp=0xc0050f6638 sp=0xc0050f6610 pc=0x252d158
github.com/tinygo-org/tinygo/builder.runJobs(0xc0012bde30?, 0xc003f63118?)
/home/rockwood/Workspace/tinygo/builder/jobs.go:132 +0x5e7 fp=0xc0050f6a30 sp=0xc0050f6638 pc=0x2872fe7
github.com/tinygo-org/tinygo/builder.Build({0x7fff5a45840c, 0x11}, {0x7fff5a458358, 0xa6}, {0xc00002e6d8, 0x15}, 0xc0001182d0)
/home/rockwood/Workspace/tinygo/builder/build.go:877 +0x369a fp=0xc0050f7538 sp=0xc0050f6a30 pc=0x286333a
main.Build({0x7fff5a45840c, 0x11}, {0x7fff5a458358, 0xa6}, 0xc0001161e0)
/home/rockwood/Workspace/tinygo/main.go:168 +0x26f fp=0xc0050f77a0 sp=0xc0050f7538 pc=0x288b2af
main.main()
/home/rockwood/Workspace/tinygo/main.go:1573 +0x3588 fp=0xc0050f7f80 sp=0xc0050f77a0 pc=0x2898fa8
runtime.main()
/usr/local/go/src/runtime/proc.go:250 +0x207 fp=0xc0050f7fe0 sp=0xc0050f7f80 pc=0x255da27
runtime.goexit()
/usr/local/go/src/runtime/asm_amd64.s:1598 +0x1 fp=0xc0050f7fe8 sp=0xc0050f7fe0 pc=0x258d8a1
goroutine 2 [force gc (idle)]:
runtime.gopark(0x0?, 0x0?, 0x0?, 0x0?, 0x0?)
/usr/local/go/src/runtime/proc.go:381 +0xd6 fp=0xc0000b0fb0 sp=0xc0000b0f90 pc=0x255de56
runtime.goparkunlock(...)
/usr/local/go/src/runtime/proc.go:387
runtime.forcegchelper()
/usr/local/go/src/runtime/proc.go:305 +0xb0 fp=0xc0000b0fe0 sp=0xc0000b0fb0 pc=0x255dc90
runtime.goexit()
/usr/local/go/src/runtime/asm_amd64.s:1598 +0x1 fp=0xc0000b0fe8 sp=0xc0000b0fe0 pc=0x258d8a1
created by runtime.init.6
/usr/local/go/src/runtime/proc.go:293 +0x25
goroutine 3 [GC sweep wait]:
runtime.gopark(0x1?, 0x0?, 0x0?, 0x0?, 0x0?)
/usr/local/go/src/runtime/proc.go:381 +0xd6 fp=0xc0000b1780 sp=0xc0000b1760 pc=0x255de56
runtime.goparkunlock(...)
/usr/local/go/src/runtime/proc.go:387
runtime.bgsweep(0x0?)
/usr/local/go/src/runtime/mgcsweep.go:319 +0xde fp=0xc0000b17c8 sp=0xc0000b1780 pc=0x254a5fe
runtime.gcenable.func1()
/usr/local/go/src/runtime/mgc.go:178 +0x26 fp=0xc0000b17e0 sp=0xc0000b17c8 pc=0x253fa66
runtime.goexit()
[... 563 lines cut ...]
tinygo.org/x/go-llvm.Builder.SetCurrentDebugLocation.func1({0xc000033830?}, 0x21f, 0x15, {0x21f?}, {0x15?})
/home/rockwood/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20221028183034-8341240c0b32/ir.go:1377 +0xb9 fp=0xc005187248 sp=0xc005187200 pc=0x2774d99
tinygo.org/x/go-llvm.Builder.SetCurrentDebugLocation({0xc005187298?}, 0xc80c6?, 0xef0000000277917b?, {0x2?}, {0xc000d25110?})
/home/rockwood/go/pkg/mod/tinygo.org/x/go-llvm@v0.0.0-20221028183034-8341240c0b32/ir.go:1377 +0x19 fp=0xc005187280 sp=0xc005187248 pc=0x2774c79
github.com/tinygo-org/tinygo/compiler.(*builder).setDebugLocation(0xc009834300, 0xc80c6)
/home/rockwood/Workspace/tinygo/compiler/compiler.go:719 +0x10e fp=0xc005187470 sp=0xc005187280 pc=0x27f25ce
github.com/tinygo-org/tinygo/compiler.(*builder).createInstruction(0xc009834300, {0x959f28?, 0xc00665d5e0?})
/home/rockwood/Workspace/tinygo/compiler/compiler.go:1455 +0x8b fp=0xc0051875b8 sp=0xc005187470 pc=0x27f7dcb
github.com/tinygo-org/tinygo/compiler.(*builder).createFunction(0xc009834300)
/home/rockwood/Workspace/tinygo/compiler/compiler.go:1349 +0xaa5 fp=0xc0051877c0 sp=0xc0051875b8 pc=0x27f7a65
github.com/tinygo-org/tinygo/compiler.(*compilerContext).createPackage(0xc003fd8280, {0xc003ed0900?}, 0xc000f92780)
/home/rockwood/Workspace/tinygo/compiler/compiler.go:865 +0x345 fp=0xc005187a28 sp=0xc0051877c0 pc=0x27f3a05
github.com/tinygo-org/tinygo/compiler.CompilePackage({0xc00031bbf4?, 0x53?}, 0xc0004381e0, 0xc000f92780, {0x7fb82621b878?}, 0x8d2d62?, 0x80?)
/home/rockwood/Workspace/tinygo/compiler/compiler.go:312 +0x459 fp=0xc005187bd0 sp=0xc005187a28 pc=0x27ef479
github.com/tinygo-org/tinygo/builder.Build.func3(0xc003d37920)
/home/rockwood/Workspace/tinygo/builder/build.go:357 +0x20e fp=0xc005187f78 sp=0xc005187bd0 pc=0x286762e
github.com/tinygo-org/tinygo/builder.runJob(0xc003d37920, 0x0?)
/home/rockwood/Workspace/tinygo/builder/jobs.go:222 +0x4f fp=0xc005187fc0 sp=0xc005187f78 pc=0x2873a4f
github.com/tinygo-org/tinygo/builder.runJobs.func2()
/home/rockwood/Workspace/tinygo/builder/jobs.go:123 +0x2a fp=0xc005187fe0 sp=0xc005187fc0 pc=0x287338a
runtime.goexit()
/usr/local/go/src/runtime/asm_amd64.s:1598 +0x1 fp=0xc005187fe8 sp=0xc005187fe0 pc=0x258d8a1
created by github.com/tinygo-org/tinygo/builder.runJobs
/home/rockwood/Workspace/tinygo/builder/jobs.go:123 +0x5be
```

deadprogram, 2023-08-22:
@rockwotj which version of TinyGo is this?

rockwotj, 2023-08-22:
Sorry it was v0.28.1

rockwotj, 2023-08-22:
Built from source using the commands to get a statically linked binary

AcousticTypewriter, 2023-08-25:
I'm having the same issue. It seems to happen a few times in a row and then disappear.

tinygo version produces: 

`tinygo version 0.28.1 linux/amd64 (using go version go1.19.8 and LLVM version 15.0.0)`

aykevl, 2023-08-29:
Are you perhaps compiling code that contains generics? I'm aware of a race condition that sometimes happens and I keep postponing to fix, see https://github.com/tinygo-org/tinygo/issues/3532.

rockwotj, 2023-08-29:
Yes my code does have generics

rockwotj, 2023-08-29:
Oh sorry wait no this version does not have any generics. I added generics after we started seeing this crash

aykevl, 2023-09-18:
This might fix the crash: https://github.com/tinygo-org/tinygo/pull/3917

deadprogram, 2023-09-21:
This is addressed as part of `v0.30.0` release so now closing. Thanks!

closed 2023-09-21 by deadprogram

rockwotj, 2023-10-27:
@deadprogram @aykevl I've confirmed that upgrading to v0.30.0 does resolve the issue (I haven't seen any crashes) thanks for your work on this!