tinygo-org/tinygo #3876: [Question] Noticeable performance differences compared to C

DrMagPie, 2023-08-19:
Hi tinyGo team!
I'm currently experimenting with tinyGo, and so far I'm loving it, big thank you for that!
And to learn tinyGo I am porting simple Arduino Nano + DS3231 (RTC) + hd44780i2c (2004 LCD display) clock. And I noticed that tinyGo takes noticeably more time to update the display.
I was expecting some performance differences, but not so big.
So my question is, am I doing something wrong? or there really is such a big difference?
Have anyone done any benchmarking co compare tinyGo to C/C++?

TinyGo: https://streamable.com/p0983z slow startup, characters appear one by one and dots between numbers appear one after another.
C: https://streamable.com/u1zxp1 fast startup, characters appear instantly, dots blink are synchronised.

deadprogram, 2023-08-19:
See https://www.mdpi.com/2079-9292/12/1/143 for a research paper on this subject.

> The results show that the C/C++ implementations were fastest in most cases, closely followed by TinyGo and Rust...

Must be something much different in your code sample, but I have not had time yet to take a look.

DrMagPie, 2023-08-19:
Thank you for sharing this paper, it's an interesting read.

I have uploaded sample code to [this](https://github.com/DrMagPie/tinyExperiment) repository. (in case you would like to have a closer look, or even run it) I'm using Arduino nano with old bootloader.

Also, please be ware this is a stripped down version of program from videos attached above. But the screen behaves exactly the same way as in the videos.

soypat, 2023-08-23:
On inspecting the tinygo driver it is clear there is room for improvement:
- Heap allocation, could be avoided by using struct allocated buffer: https://github.com/tinygo-org/drivers/blob/5e0191655b602ecc49b70aa3c0816834d85c9718/hd44780i2c/hd44780i2c.go#L217
- Could be global variable: https://github.com/tinygo-org/drivers/blob/5e0191655b602ecc49b70aa3c0816834d85c9718/hd44780i2c/hd44780i2c.go#L124C2-L124C45
- Also, I'd hope LLVM is optimizing [this conversion](https://github.com/tinygo-org/drivers/blob/5e0191655b602ecc49b70aa3c0816834d85c9718/hd44780i2c/hd44780i2c.go#L145), which is unecessary

The [C version](https://github.com/johnrickman/LiquidCrystal_I2C/blob/738765e388816fb2a687d40436cd2604febce1b1/LiquidCrystal_I2C.cpp#L269) does not heap allocate in the call to expanderWrite

Gustavomurta, 2023-10-17:
My suggestion:
 If you use a logic analyzer on the I2C bus you will have a clear perception why with TnyGo it is slower. And fix the possible problem.

_https://youtu.be/rR5cEFRO9_s?si=bT2rKCLyGnW0LhUO_

Gustavomurta, 2023-10-18:
DrMagPie, 
I was unable to test your Tinygo program on my Arduino Nano, as I had already updated the boot loader (new version).

```
PS C:\Users\jgust\tinygo\programas\arduino\tiny_experiment\tinygo_code> tinygo flash -target=arduino-nano main.go
avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude.exe: stk500_recv(): programmer is not responding

avrdude.exe: stk500_recv(): programmer is not responding
avrdude.exe: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00
avrdude.exe: ser_drain(): read error: Acesso negado.

avrdude.exe: opening programmer "arduino" on port "COM4" failed

avrdude.exe done.  Thank you.
```

So I tested your program with an Arduino Uno. With Arduino sketch it ran normally.
But with Tinygo, it showed the same symptom of slowness.

Checking the I2C bus with Logic Analyzer, I noticed that with TinyGo, the frequency of the I2C interface does not match the configuration.

 I tested with this example - _https://github.com/tinygo-org/drivers/tree/release/examples/hd44780i2c_

With 100 KHz configured - frequency detected: 1.7 KHz approximated 
With 400 KHz configured - Frequency detected - 10 KHz approximated

machine.I2C0.Configure(machine.I2CConfig{
		Frequency: 400 * machine.KHz,
	})

Tomorrow, I will analyze this problem.

Gustavomurta, 2023-10-19:
Today I tested your program with Raspberry Pico and it worked OK!_https://github.com/DrMagPie/tinyExperiment/tree/main/tinyGo_

I just made this change and the clock frequency was 95 KHz (acceptable).

`machine.I2C0.Configure(machine.I2CConfig{
		Frequency: 100 * machine.KHz, // I2C clock frequency 100 KHz
	})`

I think there is really something wrong with the I2C protocol on the Arduino TinyGo (Atmega328).

Gustavomurta, 2023-10-21:
I think i found why I2C clock frequency is wrong using Arduino TinyGo! 

Studing source code at:
_https://github.com/tinygo-org/tinygo/blob/release/src/machine/machine_atmega.go_

I think it is wrong - not SetBits, but ClearBits is correct. 
`	// Initialize twi prescaler and bit rate.
	avr.TWSR.SetBits((avr.TWSR_TWPS0 | avr.TWSR_TWPS1))`

Comparing with Arduino Sorce code:
_https://github.com/arduino/ArduinoCore-avr/blob/master/libraries/Wire/src/utility/twi.c_

`  // initialize twi prescaler and bit rate
  cbi(TWSR, TWPS0);      // cbi clear the bit 
  cbi(TWSR, TWPS1);
  TWBR = ((F_CPU / TWI_FREQ) - 16) / 2;`

[image]

Gustavomurta, 2023-10-21:
> DrMagPie, I was unable to test your Tinygo program on my Arduino Nano, as I had already updated the boot loader (new version).
> 

II discovered that there is a version for the new Arduino Nano.

tinygo targets : arduino-nano-new

`tinygo build -target=arduino-nano-new main.go`

Gustavomurta, 2023-10-22:
On the Slack Forum, I received guidance from Kenneth Bell: (Thanks Kenneth) to fix the bug on Github:

You can simply modify the local copy of the 'machine' package to see if the fix works. Just make a new version of your application.If it works, the instructions for 'building from source' are here: _https://tinygo.org/docs/guides/build/_
You can fork the tinygo repository, commit your changes and create a PR

Since I have limited experience with Github, I will try to fix it as soon as possible.

Gustavomurta, 2023-10-22:
But, I do a test with success! 

`
package main

import (
	"device/avr"
	"machine"
)

var (
	i2cFrequency uint32 = 100000
	dataTest            = []byte{0x57, 0x00, 0x01, 0x2, 0x03, 0x04}
)

func main() {

	avr.TWSR.ClearBits((avr.TWSR_TWPS0 | avr.TWSR_TWPS1))
	avr.TWBR.Set(uint8(((machine.CPUFrequency() / i2cFrequency) - 16) / 2))
	avr.TWCR.Set(avr.TWCR_TWEN)

	// Clear TWI interrupt flag, put start condition on SDA, and enable TWI.
	avr.TWCR.Set((avr.TWCR_TWINT | avr.TWCR_TWSTA | avr.TWCR_TWEN))
	//avr.TWCR.Set(avr.TWCR_TWINT)

	for i := 0; i < 7; i++ {
		writeByte(dataTest[i])
	}
}

// writeByte writes a single byte to the I2C bus.
func writeByte(data byte) error {
	// Write data to register.
	avr.TWDR.Set(data)

	// Clear TWI interrupt flag and enable TWI.
	avr.TWCR.Set(avr.TWCR_TWEN | avr.TWCR_TWINT)

	// Wait till data is transmitted.
	for !avr.TWCR.HasBits(avr.TWCR_TWINT) {
	}
	return nil

}
`

The I2C clock frequency was changed to 100 KHz ! 

[image]

Gustavomurta, 2025-09-30:
I don't know. I don't know how to suggest a fix on github.

aykevl, 2025-10-04:
You can do this roughly like this:

1. Fork the repository.
2. Clone the repository.
3. Switch to a new branch (optional).
4. Build TinyGo. See: https://tinygo.org/docs/guides/build/
5. Modify the file that needs modifications. (You don't need to build TinyGo again in this case).
6. Test that your modification fixes the bug.
7. Commit the result. Make sure to only include the changes that you want to include.
8. Push the branch to GitHub.
9. Make a PR with the changes.

I'm sure there are many tutorials out there for this, but this is roughly how it works for TinyGo.

aykevl, 2025-10-05:
Yes, you can compile on Windows. See the "Manual LLVM Build" page for that.

davecheney, 2026-10-24:
The prescaler discrepancy remains in `src/machine/machine_atmega.go`, in `I2C.SetBaudRate`: it sets both prescaler bits, while the bit-rate calculation assumes no prescaling.

soypat, 2026-10-25:
Before I unsubscribe from this thread I'd like to link the new Driver design page written by yours truly.

https://tinygo.org/docs/guides/driver-design/

It now lays out most of the rules you need to know to avoid performance hit which was (a small part compared to gustavo's findings) of this issue