tinygo-org/tinygo #3878: Issue Using Multiple Goroutines on Pi Pico

AcousticTypewriter, 2023-08-23:
Hey there!

I've been having some issues using goroutines for my raspberry pi pico. I am completely stuck from this bug. It seems there's an llvm problem compiling my code, and after some googling, I was unable to find anything related. 

Here's a simplified version of my code that still produces the error:

`package main

import (
        "machine"
        "time" 
 )

func door () {
  //run door motor
}

func elevatorYO () {
  //run elevator motor
}

func main() {
  cond := true
  for {
    result, err := machine.GetRNG()
    if err != nil {
      println("uh oh")
    } else if result == 10 {
      cond = false
    }

    switch cond {
      case true:
        go door()
      case false:
        go elevatorYO()
    }
    time.Sleep(time.Millisecond*1)
  }  
}
`
(In actuality, instead of a random number determining which goroutine is triggered, its based off of a button press)

Here's my command:
tinygo build -target=pico main.go

Here's the error:
`invalid llvm.used member
i32 ptrtoint (ptr @"main.elevatorYO$gowrapper" to i32)
LLVM ERROR: Broken module found, compilation aborted!
error: failed to link /tmp/tinygo2939992984/main: signal: aborted (core dumped)`

I'm compiling using debian.

Any ideas? I'm not sure if this is a problem using multiple goroutines or if there's something else.

I'm relatively new to tinygo and golang, so any help would be appreciated.

Cheers!

deadprogram, 2023-09-07:
I run multiple goroutines on rp2040 processors frequently, so probably not that. For example see https://github.com/hybridgroup/tinyglobo/blob/main/main.go

AcousticTypewriter, 2023-09-08:
@deadprogram I've narrowed the problem down to having multiple conditional goroutines that don't trigger consistently. That code example doesn't have that I believe.

aykevl, 2023-09-08:
I can reproduce the compiler crash.

aykevl, 2023-09-09:
This is actually a rather interesting bug. We insert a special call that is later removed again to get the maximum stack size of a function (if it can be calculated). Originally it looks like this:

```go
getGoroutineStackSize(door)
...
getGoroutineStackSize(elevatorYO)
```

...but LLVM thinks it is smart and converts it to code equivalent to this:

```go
fn := someCondition ? door : elevatorYO
getGoroutineStackSize(fn)
```

This later pass that transforms the `getGoroutineStackSize` call chokes on this transformation: it still assumes the function parameter is a function, not a variable containing a reference to a function.

Normally a transformation like this would be a good idea, but in this case it actually makes writing the pass more difficult.

I'll have to think a bit on how to correctly fix this bug.

aykevl, 2023-09-09:
I think a fix would be a rewrite of the `getGoroutineStackSize` pseudo-call and convert it to actual loads from external globals (that later get merged into one big global). I don't see how the optimizer could mess that up. It's a bit involved though.

davecheney, 2026-10-24:
A minimal program with the same conditional-goroutine pattern still crashes the compiler:

```go
package main

import (
	"machine"
	"time"
)

func first()  {}
func second() {}

func main() {
	condition := true
	for {
		value, err := machine.GetRNG()
		if err != nil {
			panic(err)
		}
		if value == 10 {
			condition = false
		}
		switch condition {
		case true:
			go first()
		case false:
			go second()
		}
		time.Sleep(time.Millisecond)
	}
}
```

Save as `main.go` and build:

```sh
tinygo build -target=pico -o main.elf main.go
```

Tested with:

```text
tinygo version 0.44.0-dev-eae3f875 darwin/arm64 (using go version go1.27.3 and LLVM version 23.1.4)
```

The compiler exits with a segmentation fault during `LLVMGoWriteThinLTOBitcodeToMemoryBuffer`. This differs from the original `invalid llvm.used member` diagnostic, so it does not establish that the underlying cause is unchanged. No firmware was run.

closed 2026-10-24 by deadprogram