tinygo-org/tinygo #3879: Prometheus not compatible with tinygo?

f0o, 2023-08-23:
Possibly related to https://github.com/tinygo-org/tinygo/issues/3705 ?

```
# github.com/prometheus/client_golang/prometheus/internal
../../../../go/pkg/mod/github.com/prometheus/client_golang@v1.16.0/prometheus/internal/go_runtime_metrics.go:42:27: d.Name undefined (type *metrics.Description has no field or method Name)
../../../../go/pkg/mod/github.com/prometheus/client_golang@v1.16.0/prometheus/internal/go_runtime_metrics.go:65:7: d.Cumulative undefined (type *metrics.Description has no field or method Cumulative)
../../../../go/pkg/mod/github.com/prometheus/client_golang@v1.16.0/prometheus/internal/go_runtime_metrics.go:65:23: d.Kind undefined (type *metrics.Description has no field or method Kind)
../../../../go/pkg/mod/github.com/prometheus/client_golang@v1.16.0/prometheus/internal/go_runtime_metrics.go:65:39: undefined: metrics.KindFloat64Histogram
../../../../go/pkg/mod/github.com/prometheus/client_golang@v1.16.0/prometheus/internal/go_runtime_metrics.go:70:11: d.Kind undefined (type *metrics.Description has no field or method Kind)
../../../../go/pkg/mod/github.com/prometheus/client_golang@v1.16.0/prometheus/internal/go_runtime_metrics.go:71:15: undefined: metrics.KindUint64
../../../../go/pkg/mod/github.com/prometheus/client_golang@v1.16.0/prometheus/internal/go_runtime_metrics.go:72:15: undefined: metrics.KindFloat64
../../../../go/pkg/mod/github.com/prometheus/client_golang@v1.16.0/prometheus/internal/go_runtime_metrics.go:73:15: undefined: metrics.KindFloat64Histogram
```

I'm not even sure where to start looking on this one; I cleared the modcache and let tinygo download the modules fresh but the error is a bit generic.

It works fine on Go 1.19+

cpanato, 2024-02-11:
go the same issue, and not sure how to fix it :/

aykevl, 2024-02-19:
The best way to get this working would be to extend the stub runtime/metrics package that got added in this PR: https://github.com/tinygo-org/tinygo/pull/3709

cpanato, 2025-08-22:
seems have a new issue

```
# github.com/prometheus/procfs
../../../../pkg/mod/github.com/prometheus/procfs@v0.16.1/fs_statfs_type.go:25:18: undefined: syscall.Statfs_t
../../../../pkg/mod/github.com/prometheus/procfs@v0.16.1/fs_statfs_type.go:26:17: undefined: syscall.Statfs
../../../../pkg/mod/github.com/prometheus/procfs@v0.16.1/proc_stat.go:214:20: undefined: os.Getpagesize
```

v1rtl, 2026-09-28:
new sets of errors on latest tinygo, all `MemStats` related:

```
# github.com/prometheus/client_golang/prometheus
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector.go:123:68: ms.StackInuse undefined (type *runtime.MemStats has no field or method StackInuse)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector.go:131:68: ms.StackSys undefined (type *runtime.MemStats has no field or method StackSys)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector.go:139:68: ms.MSpanInuse undefined (type *runtime.MemStats has no field or method MSpanInuse)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector.go:147:68: ms.MSpanSys undefined (type *runtime.MemStats has no field or method MSpanSys)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector.go:155:68: ms.MCacheInuse undefined (type *runtime.MemStats has no field or method MCacheInuse)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector.go:163:68: ms.MCacheSys undefined (type *runtime.MemStats has no field or method MCacheSys)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector.go:171:68: ms.BuckHashSys undefined (type *runtime.MemStats has no field or method BuckHashSys)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector.go:187:68: ms.OtherSys undefined (type *runtime.MemStats has no field or method OtherSys)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector.go:195:68: ms.NextGC undefined (type *runtime.MemStats has no field or method NextGC)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector_latest.go:446:5: ms.Lookups undefined (type *runtime.MemStats has no field or method Lookups)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector_latest.go:454:5: ms.StackInuse undefined (type *runtime.MemStats has no field or method StackInuse)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector_latest.go:455:5: ms.StackSys undefined (type *runtime.MemStats has no field or method StackSys)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector_latest.go:455:19: ms.StackInuse undefined (type *runtime.MemStats has no field or method StackInuse)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector_latest.go:456:5: ms.MSpanInuse undefined (type *runtime.MemStats has no field or method MSpanInuse)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector_latest.go:457:5: ms.MSpanSys undefined (type *runtime.MemStats has no field or method MSpanSys)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector_latest.go:457:19: ms.MSpanInuse undefined (type *runtime.MemStats has no field or method MSpanInuse)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector_latest.go:458:5: ms.MCacheInuse undefined (type *runtime.MemStats has no field or method MCacheInuse)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector_latest.go:459:5: ms.MCacheSys undefined (type *runtime.MemStats has no field or method MCacheSys)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector_latest.go:459:20: ms.MCacheInuse undefined (type *runtime.MemStats has no field or method MCacheInuse)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector_latest.go:460:5: ms.BuckHashSys undefined (type *runtime.MemStats has no field or method BuckHashSys)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector_latest.go:462:5: ms.OtherSys undefined (type *runtime.MemStats has no field or method OtherSys)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector_latest.go:463:5: ms.NextGC undefined (type *runtime.MemStats has no field or method NextGC)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/go_collector_latest.go:470:5: ms.GCCPUFraction undefined (type *runtime.MemStats has no field or method GCCPUFraction)
../../go/pkg/mod/github.com/prometheus/client_golang@v1.24.1/prometheus/num_threads.go:23:18: undefined: runtime.ThreadCreateProfile
```

Will make a PR to add stubs for it too

0pcom, 2026-10-20:
The MemStats errors in the last comment are fixed on this branch: https://github.com/tinygo-org/tinygo/compare/dev...0magnet:tinygo:memstats-fields. It fills in the remaining scalar MemStats fields that client_golang's Go collector reads, such as StackInuse, NextGC and GCCPUFraction, and adds ThreadCreateProfile. The pause arrays stay out, since they would add 4 KiB to a struct that microcontroller code puts on the stack.