tinygo-org/tinygo #3881: wasm-opt error parsing wasm [parse exception: invalid function index (at 0:864478)]

DarkByteBen, 2023-08-27:
[parse exception: invalid function index (at 0:864478)]
Fatal: error parsing wasm
error: wasm-opt failed: exit status 1

Setting -opt=0 would allow my build to complete, but any other opt setting including default results in a similar error.  I am not really clear on what fundamentally causes this to happen since my project is rather large and the parsing error only pointed me towards wasm-opt.  I decided to try tinkering with tinygo and found that upgrading the binaryen submodule from version 102 to 114 solved it.  Could that be upgraded for a future tinygo release?

deadprogram, 2023-08-27:
> Could that be upgraded for a future tinygo release?

Hi @DarkByteBen thanks for the info. Perhaps you would like to submit a PR with this upgrade?

ruslantalpa, 2023-12-12:
I've also encountered this issue but using a different version of wasm-opt did not work
things i observed, the compilation succeeds and main.o and main files are produced but the stage where wasm-opt is executed fails.

I should note that in my case, i "stub" a few functions in go/src and tinygo/src so that i get over a few "missing" things, however i don't think this is related to that.

when i run wasm-opt with debug, the output (at the end) is as follows

```
<==
getInt8: 0 (at 1307390)
getS64LEB: 0 ==>
zz recurse from 2 at 1307391
zz recurse into 2 at 1307391
getInt8: 32 (at 1307391)
readExpression seeing 32
zz node: LocalGet 1307392
<==
getInt8: 15 (at 1307392)
getU32LEB: 15 ==>
zz recurse from 2 at 1307393
zz recurse into 2 at 1307393
getInt8: 66 (at 1307393)
readExpression seeing 66
zz node: Const, code B
<==
getInt8: 0 (at 1307394)
getS64LEB: 0 ==>
zz recurse from 2 at 1307395
zz recurse into 2 at 1307395
getInt8: 16 (at 1307395)
readExpression seeing 16
zz node: Call
<==
getInt8: 128 (at 1307396)
getInt8: 192 (at 1307397)
getInt8: 129 (at 1307398)
getInt8: 128 (at 1307399)
getInt8: 0 (at 1307400)
getU32LEB: 24576 ==>
[parse exception: invalid function index (at 0:1307401)]
Assertion failed: (name && "Cannot print an empty name"), function printName, file Print.cpp, line 53.
[1]    74320 abort      wasm-opt --debug -o main.wasm main
```

and the line/assert in wasm-opt which seems to generate the error is this
https://github.com/WebAssembly/binaryen/blob/main/src/passes/Print.cpp#L55

Is there a way to track from where `printName` is called with a "empty" function name?

PS: i can provide privately `main` and `main.o` if that helps

deadprogram, 2024-02-28:
Completed with `v0.31.0` so now closing. Thank you!

closed 2024-02-28 by deadprogram